dusty-nv/jetson-inference #716: How to get the frame from cudaImage

john4d20, 2020-09-10:
I need to use the frame to get some variables, I discover the cudaImage and videoOutput could not return the frame.
so, i use another way to get the frame, using cv2 to get the rtsp frame.However, in the end, i could not use the cv2.imshow('frame',frame) to show the video even there is no any error showed in the terminal

john4d20, 2020-09-10:
I found that cudaImage could be converted to numpy, however, when I show the image with using cv2 rectangle, it could not show the cv2 image. Cv2 may be conflicted with the library

dusty-nv, 2020-09-10:
Hi @john4d20 , please see the `cudaToNumpy()` function from here:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays

closed 2020-09-11 by john4d20