dusty-nv/jetson-inference #718: Object Detection model performance on TensorRT

Sgomes14843, 2020-09-11:
Hi, what would be the changes to be made in the object detection eval_ssd.py script, https://github.com/dusty-nv/pytorch-ssd/tree/01e0146eb835860cac71763727530d463af67894, for him to evaluate the model optimized by TensorRT. I already did the evaluation of the pytorch model but I would like to make the model that TensorRT optimizes, to compare later. Any help is appreciated.

dusty-nv, 2020-09-11:
Hi @Sgomes14843 , to use the TensorRT version, use the `detectnet` or `detectnet.py` programs from the jetson-inference repo.

Sgomes14843, 2020-09-11:
@dusty-nv The detectnet.py script can do the same as this script https://github.com/dusty-nv/pytorch-ssd/blob/01e0146eb835860cac71763727530d463af67894/eval_ssd.py ?? Maybe I explained my question badly. Because what I want is to do what this eval_ssd script does in the pytorch model but in the tensorrt optimized model

dusty-nv, 2020-09-11:
You would need to either adapt eval_ssd to use jetson.inference.detectNet object, or integrate the dataset/label loading into detectnet.py.

Sgomes14843, 2020-09-11:
Thanks I think I'll try the first option, to see if it's possible

closed 2020-09-14 by Sgomes14843