dusty-nv/jetson-inference #719: Quick Question regarding detectnet.py and segnet.py

Rbara00, 2020-09-11:
Hi I am extremely new to using the jetson TX2, and for this project I am working on, I think the best approach to reach my team's goals is to modify the python examples, really our only issue is we're not exactly sure what we are supposed to edit to change the camera from the onboard TX2 camera to our Zedd USB. Any suggestions to understanding these two programs line by line?

Rbara00, 2020-09-11:
Nevermind it all makes sense now.

closed 2020-09-11 by Rbara00

Awais-commit, 2020-10-14:
Hi did you solve the problem? I'm trying to use my USB camera and change to "/dev/video1" but it shows the video from the onboard TX2 camera. Thanks

dusty-nv, 2020-10-14:
> Hi did you solve the problem? I'm trying to use my USB camera and change to "/dev/video1" but it shows the video from the onboard TX2 camera. Thanks

Hmm, what is the command line you are running?

If you run `detectnet /dev/video1`, it should show your USB camera...

If not, can you post the terminal output here?

Awais-commit, 2020-10-19:
> Hmm, what is the command line you are running?
>
> If you run `detectnet /dev/video1`, it should show your USB camera...
>
> If not, can you post the terminal output here?

@dusty-nv thanks for your reply. It works, I really appreciate your help and time. Thanks again.