dusty-nv/jetson-inference #720: Display video from jetson-utils with PyQt5 GUI.

lich-mk, 2020-09-15:
I want to use the GUI in my program based on jetson-inference. The GUI should perform the following functions: display the video in full screen mode, display control buttons over video, open the settings menu. I plan to use PyQt5 for this purpose.

How can I stream  video from jetson-utils to the PyQt5 (Qwidget) window (preferably without copying a frame from GPU memory to CPU memory)? As far as I understand, the gstDecoder module creates a Gstreamer pipeline with the output "...appsink name=mysink", then the display is produced by the gstDisplay module which creates its own window. 

If this is not possible with PyQt5, are there other ways to do this?

I use Jetson Nano with JP 4.4 and Python 3.6.

Thanks in advance for your answers!

lich-mk, 2020-11-10:
Hello! I solved this problem.

This task has two parts:

1. Displaying video from some Gstreamer pipline in a Qt window
2. Transfer video from jetson-inference to Gstreamer  external pipeline

Solution:

1. This code plays a test Gstreamer video (videotestsrc) in a 640x480 window.

```
import sys
from threading import Thread

from PyQt5.QtCore import *
from PyQt5.QtGui import *
from PyQt5.QtWidgets import *

import jetson.utils

import gi
gi.require_version('Gst', '1.0')
gi.require_version('GstVideo', '1.0')
from gi.repository import Gst, GObject, GstVideo
GObject.threads_init()
Gst.init(None)

class MainWindow(QMainWindow):   
    def __init__(self):
        QMainWindow.__init__(self)
        self.setAttribute(Qt.WA_AcceptTouchEvents, True)
        self.setGeometry(0,0,640,480)
        self.videowidget = VideoWidget(parent=self)
       
class VideoWidget(QWidget):   
    def __init__(self, parent):
        QMainWindow.__init__(self, parent)
        self.windowId = self.winId()
        self.setGeometry(0,0,640,480)

    def setup_pipeline(self):           
        self.pipeline = Gst.Pipeline()
        self.pipeline = "videotestsrc ! video/x-raw,width=640,height=480 ! videoconvert ! xvimagesink"
        self.pipeline = Gst.parse_launch(self.pipeline)
        bus =  self.pipeline.get_bus()
        bus.add_signal_watch()
        bus.enable_sync_message_emission()
        bus.connect('sync-message::element', self.on_sync_message)
 
    def on_sync_message(self, bus, msg):
        message_name = msg.get_structure().get_name()
        print(message_name)
        if message_name == 'prepare-window-handle':
            win_id = self.windowId
            assert win_id
            print(win_id)
            imagesink = msg.src
            imagesink.set_window_handle(win_id)
            
    def  start_pipeline(self):
        self.pipeline.set_state(Gst.State.PLAYING)

 app = QApplication([])
 window = MainWindow()
 window.videowidget.setup_pipeline()
 window.videowidget.start_pipeline()
 window.show()
 sys.exit(app.exec_())
```
 
 2.1 In order to link two pipelines in one program, it is possible to use the intervideosink/intervideosrc Gstreamer module (maybe also udpsink/udpsrc, but I didn't check it). 
 As far as I understand, image transmission using intervideosink/intervideosrc does not require additional resources for encoding/decoding, unlike udp or rtsp.
 
 Accordingly, the pipeline in the previous example should be like this:

`intervideosrc channel=v0 ! xvimagesink`

Pay attention to the parameter "channel", it allows you to connect a certain pair of pipelines and can contain letters and numbers.

2.2. In turn, jetson-inference should create an output pipeline with intervideosink module. But in the original version jetson-inference has the following output streams: RTP stream, Video file, Image file, Image sequence OpenGL window.

I modified the source code and added another type, "**intervideo://0**", which creates a pipeline with intervideosink.

Video output looks like this:

`jetson.utils.videoOutput('intervideo://0', argv=['--headless'])`

Modified jetson-inference code attached.

Added:

1. **Intervideo** output stream.

Syntax:

`jetson.utils.videoOutput('intervideo://_N_', argv=['--headless'])`

Where _N_ - channel number

Value format:  channel number - int.  Note: the channel name will be generated as vN, for example "intervideosink channel=v0"

2. Additional parameters for RTSP input stream: latency, drop-on-latency, buffer-mode.

In my experience, the value of the "latency" parameter is important for working with real-time video.

Syntaх:

`[jetson.utils.videoSource('rtsp://IP', argv=['--input-codec=h264', '--latency=100'], '--drop-on-latency=100'], '--buffer-mode=0'])]
[JI_CNVS_project_mod.zip](https://github.com/dusty-nv/jetson-inference/files/5520275/JI_CNVS_project_mod.zip)
[JI_CNVS_project_mod.zip](https://github.com/dusty-nv/jetson-inference/files/5520277/JI_CNVS_project_mod.zip)

`

Where IP - IP adress of RTSP stream, input-codec - codec (for RTSP usually h264/h265 ). These parameters are required.

Value format:  latency - int, drop-on-latency - bool, buffer-mode - int. These parameters are optional. For more details see GStreamer documentation.

Changes were made to the following files:

gstDecoder.cpp
gstEncoder.cpp
videoOptions.cpp
videoOptions.h
videoOutput.cpp
URI.cpp

Changes are marked with the comments " CNVS project modification..."

Installation procedure: replace the files, copy and install the library according to the original manual.

closed 2020-11-10 by lich-mk

mirkosprojects, 2022-05-09:
Hi, I'm trying to do the same (displaying the video from jetson-utils in a PyQt5 Widget).

> 
> Installation procedure: replace the files, copy and install the library according to the original manual.
> 
Can you explain the Installation procedure a little bit further?
I tried rebuilding my docker container and running cmake but it seems like the changes don't get added to the docker image.