dusty-nv/jetson-inference #721: Quick reference problem

john4d20, 2020-09-16:
$ sudo apt-get update
$ sudo apt-get install git cmake libpython3-dev python3-numpy
$ git clone --recursive https://github.com/dusty-nv/jetson-inference
$ cd jetson-inference
$ mkdir build
$ cd build
$ cmake ../
$ make -j$(nproc)
$ sudo make install
$ sudo ldconfig

I am writing a set up script to install jetson-inference . However, in the step of  cmake ../, it will show two screens to download models and pytorch, is there any way to press enter automatically to download the models and pytorch?

dusty-nv, 2020-09-16:
Hi john4d20, you can set BUILD_INTERACTIVE=NO (although I believe that will skip installing pytorch).  It will however download the default models.

cmake ../ -DBUILD_INTERACTIVE=NO

You could patch the behavior in install_pytorch.sh to automatically install a version instead of exiting if BUILD_INTERACTIVE=NO

dusty-nv, 2020-09-16:
> You could patch the behavior in install_pytorch.sh to automatically install a version instead of exiting if BUILD_INTERACTIVE=NO

Or you could add the pytorch install commands to your own script from here:

https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-6-0-now-available/72048