dusty-nv/jetson-inference #723: SSD-VGG16 - ONNX file doesn't work properly

hoangcuongbk80, 2020-09-16:
Hi,

I trained and tested SSD-VGG16 model successfully. I am also able to convert the model to ONNX. However the detectnet with ONNX doesn't detect anything from test images. 

Any suggestion would be appreciated.

hoangcuongbk80, 2020-09-17:
@dusty-nv Do you have any suggestion? I am using Xavier.

dusty-nv, 2020-09-17:
Hi @hoangcuongbk80 , sorry I haven't tested SSD-VGG16 myself, so I am unsure.   It was part of the upstream SSD repo that I didn't use.  VGG is an older and heavyweight architecture, so I don't intend on using it.

RGring, 2021-01-22:
For those who are facing the same problem. You need to adapt the normalization here:
https://github.com/dusty-nv/jetson-inference/blob/2e5fdafcbb706d55102dc3c2bd93885a76a4a830/c/detectNet.cpp#L729

Normalization is different than in ssd-mb1 and defined here:
https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/vision/ssd/config/vgg_ssd_config.py

closed 2023-04-27 by dusty-nv