dusty-nv/jetson-inference #724: training custom segmentation model

Yong-Hoon-In, 2020-09-17:
Hi, Thanks for your support. 
I'm trying to training segmentation model with my custom voc style segmentation dataset.
I ran jetson-inference/training/python/segmentation/train.py with arg(DIR=data path,dataset=voc)

I'm facing this error 'runtime error dataset not found or corrupted'
can I train my custom segmentation dataset with my Jetson nano?

Onixaz, 2020-09-18:
Are you trying to train on Jetson Nano? Either way, I have a complete guide on how to train a custom semantic segmentation model here [https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson](https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson) (shameless self-advertising)

dusty-nv, 2020-09-18:
Thanks @Onixaz, appreciate it - great tutorial!

Onixaz, 2020-09-18:
@dusty-nv Thanks!