dusty-nv/jetson-inference #727: How to run custom Object Detection in jesonnano on host pc?

Hwang4506, 2020-09-18:
Hello! I’m studying machine learning in Korea.
Your GitHub and YouTube videos helped me a lot, Thank you!

From your GitHub, see the link below and create a custom Object Detection model with a custom dataset.
[https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md](url)
[https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md](url)
This model worked nicely on the Jetson Nano.

I tried to run this model on my PC with rtx2070 super graphics card for higher speed and higher FPS.
As much as possible, the PC environment was configured as similar to Jetson Nano.
OS : Ubuntu 18.04 LTS,     nvidia driver : 450.66,     CUDA : 10.2,     cuDNN : 7.6.5,    TensorRT : 7.1,  
Python : 3.6.9,     PyTorch : 1.3.1,       DeepStream SDK : 5.0
and I installed Jetson Inference and build, make and configured the same as I did on Jetson Nano.

Finally when I ran the model, the following error occurred
[image]
It seems to be a problem with nvjpegcodec.
I also checked the link below to fix the error, but couldn't fix it.
[https://github.com/dusty-nv/jetson-inference/issues/35](url)

Please tell us a specific way to solve it.
Finally, thank you for reading the question I wrote with poor English skills and look forward to your answer.
Thank you!

dusty-nv, 2020-09-18:
Hi @Hwang4506 , the PC does not have the GStreamer hw-accelerated codec elements like Jetson does (i.e. `nvjpecdec`).

What might work is if you run the program with `--input-codec=raw`, so no decoder needs to be used.

It is also possible on PC that there could be some memory issue, because for Jetson I am always using CUDA mapped memory without any `cudaMemcpy()` (i.e. zero-copy)

Hwang4506, 2020-09-21:
Hi, Thanks to this, the decoder problem was solved. Thank you!

when running again by adding the `--input-codec=raw` option, 
it seemed to run well, but the following new problem occurred.

The first and subsequent error messages are different,
Error message when first run is below
[image]

Error message when second and subsequent run is below.
[image]

Is there any way to solve the problem?

dusty-nv, 2020-09-21:
Ah, since you are on desktop, you need to add your GPU card's compute architecture to CMakeLists.txt and re-compile:

https://github.com/dusty-nv/jetson-inference/blob/12061e0a778ddc237bc153f81465488fe2539742/CMakeLists.txt#L63

Hwang4506, 2020-09-22:
> Ah, since you are on desktop, you need to add your GPU card's compute architecture to CMakeLists.txt and re-compile:
> 
> https://github.com/dusty-nv/jetson-inference/blob/12061e0a778ddc237bc153f81465488fe2539742/CMakeLists.txt#L63

Thank you for your answer.

After checking the device query, I added `-gencode arch=compute_61,code=sm_61`, my GPU card's compute architecture, to CMakeList.txt as follows and re-compile.
[image]

And I ran the model again but got the same error.
There seems to be a lot of difficulty in porting the model running on JetsonNano to PC.

Is there any other solution?
Thank you for your hard work.

dusty-nv, 2020-09-22:
Did you try re-running `cmake ../`, `make`, and `sudo make install`?

If you are still getting the same error, try removing your build directory and re-building from scratch.

Hwang4506, 2020-09-25:
> Did you try re-running `cmake ../`, `make`, and `sudo make install`?
> 
> If you are still getting the same error, try removing your build directory and re-building from scratch.

Hi, After seeing the answer, I deleted jetson-inference and installed it from scratch.

I downloaded a new one from GitHub and ran `cmake ../`, `make`, `sudo make install` again.
And the model was newly trained in the PC environment, the onnx file was extracted, and the engine was made.

I ran the model again, and it finally ran fine!
All thanks to you, Thank you very much!

However, I have one question.
In Jetson Nano, the fps value displayed at the top of the program was in the 40s, but on the PC, the fps value rose to 1100s. Look at the picture below
[image]

Is this the right figure? I am curious about the principle of calculating fps in a program.
And can I find the mAP value?
Thanks again for your kind and accurate answer!

closed 2020-09-25 by Hwang4506