dusty-nv/jetson-inference #728: detectnet.cpp and imagenet.cpp example scripts does not output image

Shane555, 2020-09-18:
Hello i tried running ./detectnet --network=ssd-mobilenet-v2 images/peds_0.jpg output.jpg and the window output seems to pop up briefly and closes. I cannot save the image file to output. The same thing is happening for ./imagenet images/orange_0.jpg images/test/output_0.jpg . Do you know what is going on? This is my log for ./detectnet

------------------------------------------------
imageLoader video options:
------------------------------------------------
  -- URI: file:///usr/local/jetson-inference/build/aarch64/bin/images/peds_0.jpg
     - protocol:  file
     - location:  images/peds_0.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     input
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[video]  created imageWriter from file:///usr/local/jetson-inference/build/aarch64/bin/output.jpg
------------------------------------------------
imageWriter video options:
------------------------------------------------
  -- URI: file:///usr/local/jetson-inference/build/aarch64/bin/output.jpg
     - protocol:  file
     - location:  output.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     output
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1920
  -- height:     1080
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]    TensorRT version 6.0.1
[TRT]    loading NVIDIA plugins...
[TRT]    Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]    Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]    Plugin Creator registration succeeded - NMS_TRT
[TRT]    Plugin Creator registration succeeded - Reorg_TRT
[TRT]    Plugin Creator registration succeeded - Region_TRT
[TRT]    Plugin Creator registration succeeded - Clip_TRT
[TRT]    Plugin Creator registration succeeded - LReLU_TRT
[TRT]    Plugin Creator registration succeeded - PriorBox_TRT
[TRT]    Plugin Creator registration succeeded - Normalize_TRT
[TRT]    Plugin Creator registration succeeded - RPROI_TRT
[TRT]    Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]    Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]    detected model format - UFF  (extension '.uff')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.6001.GPU.FP16.engine
[TRT]    loading network plan from engine cache... networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.6001.GPU.FP16.engine
[TRT]    device GPU, loaded networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]    Deserialize required 4540013 microseconds.
[TRT]    
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       117
[TRT]       -- maxBatchSize 1
[TRT]       -- workspace    0
[TRT]       -- deviceMemory 13220352
[TRT]       -- bindings     3
[TRT]       binding 0
                -- index   0
                -- name    'Input'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  3
                -- dim #0  3 (SPATIAL)
                -- dim #1  300 (SPATIAL)
                -- dim #2  300 (SPATIAL)
[TRT]       binding 1
                -- index   1
                -- name    'NMS'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  3
                -- dim #0  1 (SPATIAL)
                -- dim #1  100 (SPATIAL)
                -- dim #2  7 (SPATIAL)
[TRT]       binding 2
                -- index   2
                -- name    'NMS_1'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  3
                -- dim #0  1 (SPATIAL)
                -- dim #1  1 (SPATIAL)
                -- dim #2  1 (SPATIAL)
[TRT]    
[TRT]    binding to input 0 Input  binding index:  0
[TRT]    binding to input 0 Input  dims (b=1 c=3 h=300 w=300) size=1080000
[TRT]    binding to output 0 NMS  binding index:  1
[TRT]    binding to output 0 NMS  dims (b=1 c=1 h=100 w=7) size=2800
[TRT]    binding to output 1 NMS_1  binding index:  2
[TRT]    binding to output 1 NMS_1  dims (b=1 c=1 h=1 w=1) size=4
[TRT]    
[TRT]    device GPU, networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff initialized.
[TRT]    W = 7  H = 100  C = 1
[TRT]    detectNet -- maximum bounding boxes:  100
[TRT]    detectNet -- loaded 91 class info entries
[TRT]    detectNet -- number of object classes:  91
[image] loaded 'images/peds_0.jpg'  (1920x1080, 3 channels)
4 objects detected
detected obj 0  class #1 (person)  confidence=0.984155
bounding box 0  (1073.998291, 108.557114)  (1396.668335, 1003.399353)  w=322.670044  h=894.842224
detected obj 1  class #1 (person)  confidence=0.698432
bounding box 1  (458.731140, 122.974380)  (741.712280, 943.486572)  w=282.981140  h=820.512207
detected obj 2  class #1 (person)  confidence=0.975577
bounding box 2  (741.996948, 160.915314)  (1023.324951, 941.105957)  w=281.328003  h=780.190674
detected obj 3  class #1 (person)  confidence=0.863161
bounding box 3  (1478.159302, 149.118286)  (1720.584351, 915.414917)  w=242.425049  h=766.296631
[OpenGL] creating 1920x1080 texture (GL_RGB8 format, 6220800 bytes)
[cuda]   registered openGL texture for interop access (1920x1080, GL_RGB8, 6220800 bytes)
[image] failed to save 1920x1080 image to 'output.jpg'
[image] imageWriter -- failed to save 'output.jpg'

[TRT]    ------------------------------------------------
[TRT]    Timing Report networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.13369ms  CUDA   0.78243ms
[TRT]    Network       CPU  35.72004ms  CUDA  34.84893ms
[TRT]    Post-Process  CPU   0.25775ms  CUDA   0.25728ms
[TRT]    Visualize     CPU  36.25151ms  CUDA  36.87494ms
[TRT]    Total         CPU  72.36300ms  CUDA  72.76358ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[image] imageLoader -- End of Stream (EOS) has been reached, stream has been closed
detectnet:  shutting down...
detectnet:  shutdown complete.
`
`

Shane555, 2020-09-18:
Turns out it has something to do with having to use sudo due to root priviliges i think.

closed 2020-09-18 by Shane555

tonylau2010, 2022-01-19:
Thanks,it works!