dusty-nv/jetson-inference #729: Is there any way to feed Keras files into these modules?

Rbara00, 2020-09-18:
Hi I am currently working in a research lab and for our object detection, we're trying to use a TX2 and things like DetectNet and imagenet, but we have a separate team that created some custom models using Keras and we would probably want to pass them in for real-time object detection, what would be the best way of going about this? We could convert their models into TensorRT models, but how would we feed in their code as datasheets to then run these programs?

dusty-nv, 2020-09-18:
I haven't used Keras models before - can it export to ONNX?

To support different kinds of models, you typically need to add pre/post-processing code (i.e. into `c/detectNet.cpp`) that formats the input tensors appropriately (and as needed, applying mean pixel subtraction coefficients, normalization/std deviation, BGR vs RGB, ect) and then interpreting the output tensors (and doing clustering for detections)

Rbara00, 2020-09-21:
If we export it to ONNX, would it be the same as passing in a custom datasheet like you stated in the github?

dusty-nv, 2020-09-21:
The command-line options may be the same, but you may need to add customized code into detectNet.cpp depending on if the input/output tensor formats are the same as they are with the PyTorch ONNX detection models (SSD-Mobilenet).   Here is the pre/post-processing code:

https://github.com/dusty-nv/jetson-inference/blob/12061e0a778ddc237bc153f81465488fe2539742/c/detectNet.cpp#L729
https://github.com/dusty-nv/jetson-inference/blob/12061e0a778ddc237bc153f81465488fe2539742/c/detectNet.cpp#L815

Rbara00, 2020-09-21:
Assuming it would be a similar process if we were using the python version of detectnet?

Rbara00, 2020-09-28:
So the models given to me have already been converted to .UFF files for TensorRT. How could I skip the ONNX conversion and continue to feed in this model?

closed 2020-09-28 by Rbara00