dusty-nv/jetson-inference #730: How to re-train Segnet ?

Jingchensun, 2020-09-19:
I saw you have released Classification/Recognition (ResNet-18) and Object Detection (SSD-Mobilenet), but how about re-training of Segmentation Networks, I want use the segnet to train my own dataset, how should I do?

dusty-nv, 2020-09-19:
Hi @Jingchensun , I recommend checking this post:   https://github.com/dusty-nv/jetson-inference/issues/724#issuecomment-694832441

Segmentation networks are heavy and the datasets typically large, so may need to train on PC / server with discrete GPU instead.  If you use a small custom dataset like in that post I linked to, you may be able to do it on Jetson though.

Jingchensun, 2020-09-23:
> Hi @Jingchensun , I recommend checking this post: [#724 (comment)](https://github.com/dusty-nv/jetson-inference/issues/724#issuecomment-694832441)
> 
> Segmentation networks are heavy and the datasets typically large, so may need to train on PC / server with discrete GPU instead. If you use a small custom dataset like in that post I linked to, you may be able to do it on Jetson though.

Hi, thank you for your suggestion, I re-trained the segmentation model on my server successfully with your instruction.

JTDeve, 2021-09-20:
Hi,Can you teach me how to do that,thanks! @Jingchensun

Yhaw, 2023-03-24:
Please what are the steps to retrain the segnet model

dusty-nv, 2023-03-26:
Hi @Yhaw, please see this post - https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson