dusty-nv/jetson-inference #731: I wonder that why this model fps is lower than 'jetson becnhmarks' model?

hjlee9182, 2020-09-21:
[image]

In this image from https://developer.nvidia.com/embedded/jetson-benchmarks

I'm running this repo and check model fps.
ssd-mobilenet-v1( this repo)'s fps is almost 10 , But that image ssd-mobilenet-v1's fps is almost 40.

Is it normal situation? if not how to develop to high fps?

dusty-nv, 2020-09-21:
This is 90 class MS COCO model, whereas the other one was using fewer classes (37 I think)

With SSD-Mobilenet-v2 (COCO) you should get 22-25 FPS on Nano.

[image]<https://user-images.githubusercontent.com/46337218/93728999-82eba380-fbfd-11ea-9bf2-96e241117548.png>

hjlee9182, 2020-09-21:
@dusty-nv 
Now I'm running on MAXN.
any other control to develop that?

closed 2020-09-21 by hjlee9182