dusty-nv/jetson-inference #733: Question about mobilenet-v1-ssd-mp-0_675.pth

silent-code, 2020-09-21:
Does anyone know which classes this network is pre-trained on? I understand that the original ssd-mobilenet-v1 is pre-trained on the 80 classes of the COCO dataset. However this repo documentation [states](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md) that the base net we are using, **mobilenet-v1-ssd-mp-0_675.pth**, is pre-trained on the PASCAL VOC dataset. 

Is **mobilenet-v1-ssd-mp-0_675.pth** pre-trained on COCO and then fine-tuned on PASCAL VOC?

Thanks in advance!

dusty-nv, 2020-09-21:
The SSD-Mobilenet-v1 (COCO) that comes with jetson-inference i pre-trained differently i believe than the models from pytorch_ssd.

The SSD-Mobilenet-v1 (COCO) was a TensorFlow model.  Whereas I use this pytorch_ssd fork with pytorch.  I think it was pretrained on VOC as you pointed out.

silent-code, 2020-09-21:
Ok thanks Dusty. With that in mind, how can I get my retrained network (fine tuned on my custom dataset) to output my custom label as well as a subset of the original PASCAL VOC labels. So for example if I fine tuned our base network on Bananas and wanted to detect additionally on the PASCAL VOC class of Person, do I simply include Person in the labels.txt file like so?

BACKGROUND
Bananas 
Person

 I've tried this and don't appear to get Bananas and Persons, only Bananas.

Thanks!

dusty-nv, 2020-09-21:
At minimum you would need to have all the Pascal VOC classes (in their original order in labels.txt) and then add your additional class at the end, and re-train on your additional class.   I'm not sure that will work, however.   It would probably be easier to select the subset of data you want from Pascal VOC, combine it with your custom data, and then re-train the network on that dataset.

silent-code, 2020-09-22:
Thanks again for the awesome support Dusty!

closed 2020-09-22 by silent-code