dusty-nv/jetson-inference #736: Possible problem with the onnx representation

silent-code, 2020-09-22:
To test the onnx intermediate representation, I converted our original base model, **mobilenet-v1-ssd-mp-0_675.pth** to onnx with the appropriate command:

python3 onnx_export.py --input="./models/model_orig/mobilenet-v1-ssd-mp-0_675.pth" --model-dir=models/model_orig

and included the original PASCAL VOC labels in label.txt, including the BACKGROUND class, since these are the classes it was trained on [here](https://github.com/qfgaohao/pytorch-ssd).

The problem is that this network will then classify everything as class 1, or bicycle, on any input image or video sequence. This issue also exists when re-training the base net **mobilenet-v1-ssd-mp-0_675.pth** on a custom dataset. I believe @JaredP94 is experiencing a similar [issue](https://github.com/dusty-nv/jetson-inference/issues/715) as well.

Any advice? 

Thanks again, in advance.
Regards, Tim

closed 2020-09-22 by silent-code

JaredP94, 2020-09-22:
Hi @silent-code, curious to know what your issue was? :)

silent-code, 2020-09-22:
@JaredP94 

I seem to be having a similar issue as you where everything with my re-trained network classifies with the class 1 label from the list contained in label.txt:

BACKGROUND
class 1 

I thought maybe something was getting messed up in the onnx conversion step so I tested simply converting the base network, **mobilenet-v1-ssd-mp-0_675.pth** to onnx without doing any re-trainning. It appears to work correctly, classifying multiple classes with expected accuracy. Thus it appears the re-training pipleline should in theory work. This is the issue I resolved.

My next step is to just try using a larger training data set so it learns "not class 1" better, I suppose. Right now, I only have around 100 samples.

Have you made any progress?

JaredP94, 2020-10-07:
Hi @silent-code,

I didn't make any progress on mb1-ssd, but seemed to get everything working using mb2-ssd-lite (more detail [here](https://github.com/dusty-nv/jetson-inference/issues/715))