dusty-nv/jetson-inference #737: Segnet failed to load network after re-train the segmentation networks

Jingchensun, 2020-09-23:
I re-trained the segmentation network as the structions in #724 , however there is only four choice for the backbone ( 'deeplabv3_resnet101', 'deeplabv3_resnet50', 'fcn_resnet101', 'fcn_resnet50'), I then choosed fcn_resnet50 to train my dataset.  
After I converted the  best-model  trained to the ONNX model, however, my Jetson Nano seems to could not load the onnx model. Here is the compile:

`./segnet.py --camera=/dev/video0 --model=/home/jingchen/jetson-inference/examples/segnet/fcn_resnet50.onnx --width=640 --height=480 --labels=/home/jingchen/jetson-inference/examples/segnet/classes.txt --colors=/home/jingchen/jetson-inference/examples/segnet/colors.txt --input_blob="input_0" --output_blob="output_0"`

And here is the bug:
`segNet -- loading segmentation network model from:
       -- prototxt:   (null)
       -- model:      /home/jingchen/jetson-inference/examples/segnet/fcn_resnet50.onnx
       -- labels:     /home/jingchen/jetson-inference/examples/segnet/classes.txt
       -- colors:     /home/jingchen/jetson-inference/examples/segnet/colors.txt
       -- input_blob  'input_0'
       -- output_blob 'output_0'
       -- batch_size  1

[TRT]    TensorRT version 7.1.3
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - ONNX  (extension '.onnx')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file /home/jingchen/jetson-inference/examples/segnet/fcn_resnet50.onnx.1.1.7103.GPU.FP16.engine
[TRT]    cache file not found, profiling network model on device GPU
[TRT]    device GPU, loading /usr/bin/ /home/jingchen/jetson-inference/examples/segnet/fcn_resnet50.onnx
----------------------------------------------------------------
Input filename:   /home/jingchen/jetson-inference/examples/segnet/fcn_resnet50.onnx
ONNX IR version:  0.0.6
Opset version:    9
Producer name:    pytorch
Producer version: 1.6
Domain:           
Model version:    0
Doc string:       
----------------------------------------------------------------
[TRT]    Plugin creator already registered - ::GridAnchor_TRT version 1
[TRT]    Plugin creator already registered - ::NMS_TRT version 1
[TRT]    Plugin creator already registered - ::Reorg_TRT version 1
[TRT]    Plugin creator already registered - ::Region_TRT version 1
[TRT]    Plugin creator already registered - ::Clip_TRT version 1
[TRT]    Plugin creator already registered - ::LReLU_TRT version 1
[TRT]    Plugin creator already registered - ::PriorBox_TRT version 1
[TRT]    Plugin creator already registered - ::Normalize_TRT version 1
[TRT]    Plugin creator already registered - ::RPROI_TRT version 1
[TRT]    Plugin creator already registered - ::BatchedNMS_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Plugin creator already registered - ::CropAndResize version 1
[TRT]    Plugin creator already registered - ::DetectionLayer_TRT version 1
[TRT]    Plugin creator already registered - ::Proposal version 1
[TRT]    Plugin creator already registered - ::ProposalLayer_TRT version 1
[... 394 lines cut ...]
[TRT]    ModelImporter.cpp:103: Parsing node: Concat_191 [Concat]
[TRT]    ModelImporter.cpp:119: Searching for input: 512
[TRT]    ModelImporter.cpp:119: Searching for input: 517
[TRT]    ModelImporter.cpp:125: Concat_191 [Concat] inputs: [512 -> (2)], [517 -> (2)], 
[TRT]    ImporterContext.hpp:141: Registering layer: Concat_191 for ONNX node: Concat_191
[TRT]    ImporterContext.hpp:116: Registering tensor: 518 for ONNX tensor: 518
[TRT]    ModelImporter.cpp:179: Concat_191 [Concat] outputs: [518 -> (4)], 
[TRT]    ModelImporter.cpp:103: Parsing node: Upsample_192 [Upsample]
[TRT]    ModelImporter.cpp:119: Searching for input: 508
[TRT]    ModelImporter.cpp:119: Searching for input: 518
[TRT]    ModelImporter.cpp:125: Upsample_192 [Upsample] inputs: [508 -> (1, 2, 40, 40)], [518 -> (4)], 
ERROR: builtin_op_importers.cpp:3460 In function importUpsample:
[8] Assertion failed: scales_input.is_weights()
[TRT]    failed to parse ONNX model '/home/jingchen/jetson-inference/examples/segnet/fcn_resnet50.onnx'
[TRT]    device GPU, failed to load /home/jingchen/jetson-inference/examples/segnet/fcn_resnet50.onnx
[TRT]    segNet -- failed to load.
jetson.inference -- segNet failed to load network
Traceback (most recent call last):
  File "segnet.py", line 56, in <module>
    net = jetson.inference.segNet(opt.network, sys.argv)
Exception: jetson.inference -- segNet failed to load network
`
Is this because the fcn_resnet50 models too heavy that Jetson Nano can't load ?

**And By the Way, as mentioned in #437 , could you help me how to download the default resnet-based FCN models since the [mirror downloads](https://github.com/dusty-nv/jetson-inference/releases) only provide the alexnet-based FCN models**

dusty-nv, 2020-09-23:
Hi @Jingchensun , my fork of torchvision (0.3.0) branch contains the support for FCN-ResNet18 and also modifications to make it able to be loaded in TensorRT:

https://github.com/dusty-nv/vision/tree/v0.3.0

There is also a segmentation training tutorial here:   https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson

dusty-nv, 2020-09-23:
I've also uploaded the FCN-ResNet18 models to the mirror page here:   https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618

Onixaz, 2020-09-23:
@Jingchensun if you followed my tutorial, did you use "pip install git+https://github.com/Onixaz/vision@v0.3.0" to install torchvision library ? 

@dusty-nv I actually made a fix in your branch, mentioned in here https://github.com/dusty-nv/pytorch-segmentation/issues/2#issuecomment-576907769

Could make a pull request, but then again I'm not sure what causes this....

dusty-nv, 2020-09-23:
What I would like to do in the longer term is to take our customizations from torchvision and just move them into pytorch-segmentation, so that pytorch-segmentation repo does not require the customized fork of torchvision.  

Happy to team up on it @Onixaz if you are interested, although I am busy with other tasks the next few weeks.  Ideally I would like to include segmentation training tutorial in Hello AI World, but I don't know if it would run on Jetson or not due to memory/compute resources.

Onixaz, 2020-09-23:
That would be ideal. Sure @dusty-nv , I'm on a strict schedule myself, but I'll start looking into it.

marvision-ai, 2020-10-06:
@Onixaz and @dusty-nv  Thank you very much for the segmentation repo and the great tutorial! 

I am on the last step of training the network (prepped data and everything) and run into this issue. Can you pinpoint why? My images are of size `1500x624`

```
$python train.py /home/ai/algorithms/jetson_segmentation/pytorch-segmentation/5Y2_final --dataset=custom --classes=1 --width 1500 --height 624

pytorch-segmentation/datasets/__init__.py
Not using distributed mode
Namespace(arch='fcn_resnet18', aux_loss=False, batch_size=4, classes=1, data='/home/ai/algorithms/jetson_segmentation/pytorch-segmentation/5Y2_final', dataset='custom', device='cuda', dist_url='env://', distributed=False, epochs=30, height=624, lr=0.01, model_dir='.', momentum=0.9, pretrained=False, print_freq=10, resolution=320, resume='', test_only=False, weight_decay=0.0001, width=1500, workers=16, world_size=1)
=> training with dataset: 'custom' (train=460, val=116)
=> training with resolution: 1500x624, 1 classes
=> training with model: fcn_resnet18
torchvision.models.segmentation.fcn_resnet18()
torchvision.models.segmentation.FCN() => configuring model for training
Traceback (most recent call last):
  File "train.py", line 334, in <module>
    main(args)
  File "train.py", line 295, in main
    train_one_epoch(model, criterion, optimizer, data_loader, lr_scheduler, device, epoch, args.print_freq)
  File "train.py", line 192, in train_one_epoch
    loss.backward()
  File "/home/.virtualenvs/jetson_seg/lib/python3.6/site-packages/torch/tensor.py", line 107, in backward
    torch.autograd.backward(self, gradient, retain_graph, create_graph)
  File "/home/.virtualenvs/jetson_seg/lib/python3.6/site-packages/torch/autograd/__init__.py", line 93, in backward
    allow_unreachable=True)  # allow_unreachable flag
RuntimeError: CUDA error: device-side assert triggered
```
Let me know! Thank you in advance :smile:

Onixaz, 2020-10-06:
Hi @marvision-ai,

Try --classes=2, since **background** counts as a "class" as well.

marvision-ai, 2020-10-06:
Hi @Onixaz ,
Yup that was it! I feel so silly. Thanks for the help :+1:

closed 2023-04-27 by dusty-nv

khsafkatamin, 2024-03-04:
Sorry, it was a silly issue. The issue popped up as I tried to load the ONNX file generated in my PC(x86) interface into Jetson. It got sorted when I generated the ONNX format directly on Jetson.

Thanks anyways and great work!