dusty-nv/jetson-inference #739: Failed to run imagenet example

jasvinderkhurana, 2020-09-23:
Hi,

I have followed a step by step guide to work on jetson-inference examples and when I am trying to execute any imagenet example I get error like below:

`jasvinder@jasvinder-desktop:~/jassi/jetson-inference/build/aarch64/bin$ ./imagenet-console --headless images/orange_0.jpg outpu.jpg
[video]  created imageLoader from file:///home/jasvinder/jassi/jetson-inference/build/aarch64/bin/images/orange_0.jpg
------------------------------------------------
imageLoader video options:
------------------------------------------------
  -- URI: file:///home/jasvinder/jassi/jetson-inference/build/aarch64/bin/images/orange_0.jpg
     - protocol:  file
     - location:  images/orange_0.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     input
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[video]  created imageWriter from file:///home/jasvinder/jassi/jetson-inference/build/aarch64/bin/outpu.jpg
------------------------------------------------
imageWriter video options:
------------------------------------------------
  -- URI: file:///home/jasvinder/jassi/jetson-inference/build/aarch64/bin/outpu.jpg
     - protocol:  file
     - location:  outpu.jpg
     - extension: jpg
  -- deviceType: file
  -- ioType:     output
  -- codec:      unknown
  -- width:      0
  -- height:     0
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   1

[TRT]    TensorRT version 7.1.3
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - caffe  (extension '.caffemodel')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file networks/bvlc_googlenet.caffemodel.1.1.7103.GPU.FP16.engine
[TRT]    loading network plan from engine cache... networks/bvlc_googlenet.caffemodel.1.1.7103.GPU.FP16.engine
[TRT]    invalid engine cache file size (0 bytes)  networks/bvlc_googlenet.caffemodel.1.1.7103.GPU.FP16.engine
[TRT]    failed to load networks/bvlc_googlenet.caffemodel
[TRT]    imageNet -- failed to initialize.
imagenet:  failed to initialize imageNet

`

Any Idea  what is wrong here?

dusty-nv, 2020-09-23:
Can you try deleting `jetson-inference/data/networks/bvlc_googlenet.caffemodel.1.1.7103.GPU.FP16.engine` and running again?

jasvinderkhurana, 2020-09-23:
Thanks!

It worked.. 👍

closed 2020-09-23 by jasvinderkhurana