dusty-nv/jetson-inference #740: Detectnet using mjpeg pixel format problem

han-bun, 2020-09-24:
Hi everyone,

I am learning about jetson nano and is currently trying detectnet.
I am using C920 logitech camera.
When I am using detectnet using YUYV pixelformat, it runs smoothly and it does not crash.
My problem is that when I am using the mjpeg pixel format, it randomly crashes.

As far as I can see it, the maximum safe resolution that I can run the detectnet smoothly is when I set it to 800x448 which uses YUY pixel format.
But when I set it size higher than that,  pixel format becomes mjpeg and program will randomly crashes.

I also wanted to run the dectectnet on 30fps but with c920 camera, I can only do that on 1080p and the problem is that 1080p runs only on mjpeg pixel format.

Question:
1. How to fixed the detectnet crashes when using mjpeg pixel format.
2. Is it possible to run detectnet on 1080p yuyv pixel format? 

By the way, my c920 only supports YUYV and MJPEG format.

Thank you for your patience and help.

dusty-nv, 2020-09-24:
Can you try changing line 181 of `jetson-inference/utils/camera/gstCamera.cpp` to use `jpegdec` instead of `nvjpegdec`?

https://github.com/dusty-nv/jetson-utils/blob/916023814a1ac1401380c5d4313454c78c0a5f40/camera/gstCamera.cpp#L181

should become:
			`ss << "jpegdec ! video/x-raw ! ";`

Then re-run `make` and `sudo make install`.   Let me know if that helps.

> By the way, my c920 only supports YUYV and MJPEG format.

My C920 supports H264, MJPEG, and YUV, so that is odd yours does not have H264.   The H264 is really stable, although MJPEG does work for me too.

han-bun, 2020-09-25:
@dusty-nv 

Changing nvjegdec to jpedgect solved the issue. Thank you for you help.

Yes my C920 is a bit weird. When I run v4l2-ctl --list-formats, I can only YUYV and MJPEG while on other they have H264 included on their format list.

If it is ok, I have one more question.
My C920 is projecting mirror image so I wanted to horizontal flip it.
For CSI camera,  if  I want to horizontal flip the camera I will just change the value of line 144 of jetson-inference/utils/camera/gstCamera.cpp.
For v4l2, how do I that?
When I run v4l2-ctl -d /dev/video0 --all, there is no flipping option that is included.

Once again, I am sorry for asking such question and thank you for your help.

closed 2023-04-27 by dusty-nv