dusty-nv/jetson-inference #741: Re-training SSD-Mobilenet EOFError: Ran out of input

yfc1996, 2020-09-24:
There was an error that when I re-training SSD-Mobilenet,there was an error out put
nvidia@nvidia-desktop:~/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --data=data/fruit --model-dir=models/fruit --batch-size=4 --epochs=30
2020-09-24 22:30:35 - Using CUDA...
2020-09-24 22:30:35 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/fruit', dataset_type='open_images', datasets=['data/fruit'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=30, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2020-09-24 22:30:35 - Prepare training datasets.
2020-09-24 22:30:35 - loading annotations from: data/fruit/sub-train-annotations-bbox.csv
2020-09-24 22:30:35 - annotations loaded from:  data/fruit/sub-train-annotations-bbox.csv
num images:  5145
2020-09-24 22:30:47 - Dataset Summary:Number of Images: 5145
Minimum Number of Images for a Class: -1
Label Distribution:
	Apple: 3622
	Banana: 1574
	Grape: 2560
	Orange: 6186
	Pear: 757
	Pineapple: 534
	Strawberry: 7553
	Watermelon: 753
2020-09-24 22:30:47 - Stored labels into file models/fruit/labels.txt.
2020-09-24 22:30:47 - Train dataset size: 5145
2020-09-24 22:30:47 - Prepare Validation datasets.
2020-09-24 22:30:47 - loading annotations from: data/fruit/sub-test-annotations-bbox.csv
2020-09-24 22:30:47 - annotations loaded from:  data/fruit/sub-test-annotations-bbox.csv
num images:  930
2020-09-24 22:30:49 - Dataset Summary:Number of Images: 930
Minimum Number of Images for a Class: -1
Label Distribution:
	Apple: 329
	Banana: 132
	Grape: 446
	Orange: 826
	Pear: 107
	Pineapple: 105
	Strawberry: 754
	Watermelon: 125
2020-09-24 22:30:49 - Validation dataset size: 930
2020-09-24 22:30:49 - Build network.
2020-09-24 22:30:49 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
Traceback (most recent call last):
  File "train_ssd.py", line 309, in <module>
    net.init_from_pretrained_ssd(args.pretrained_ssd)
  File "/home/nvidia/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 119, in init_from_pretrained_ssd
    state_dict = torch.load(model, map_location=lambda storage, loc: storage)
  File "/home/nvidia/.local/lib/python3.6/site-packages/torch/serialization.py", line 585, in load
    return _legacy_load(opened_file, map_location, pickle_module, **pickle_load_args)
  File "/home/nvidia/.local/lib/python3.6/site-packages/torch/serialization.py", line 755, in _legacy_load
    magic_number = pickle_module.load(f, **pickle_load_args)
EOFError: Ran out of input

I use Jetson xavier NX
I hope someboy could help me

dusty-nv, 2020-09-24:
Hmm it appears that your `mobilenet-v1-ssd-mp-0_675.pth` is corrupted - can you try downloading it again?

yfc1996, 2020-09-24:
Thank you for helping me.I tried as what you did in the topic Re-training SSD-Mobilenet.But there are something wrong when I download that like this 
ps://nvidia.box.com/shared/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth -O models/mobilenet-v1-ssd-mp-0_675.pth
--2020-09-24 22:56:49--  https://nvidia.box.com/shared/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth
Resolving nvidia.box.com (nvidia.box.com)... 103.116.4.197
Connecting to nvidia.box.com (nvidia.box.com)|103.116.4.197|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: /public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth [following]
--2020-09-24 22:56:50--  https://nvidia.box.com/public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth
Reusing existing connection to nvidia.box.com:443.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://nvidia.app.box.com/public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth [following]
--2020-09-24 22:56:51--  https://nvidia.app.box.com/public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth
Resolving nvidia.app.box.com (nvidia.app.box.com)... 8.7.198.45, 2001::6ca0:a26d
Connecting to nvidia.app.box.com (nvidia.app.box.com)|8.7.198.45|:443... failed: Connection refused.
Connecting to nvidia.app.box.com (nvidia.app.box.com)|2001::6ca0:a26d|:443... failed: Network is unreachable.

yfc1996, 2020-09-24:
> 
> 
> Hmm it appears that your `mobilenet-v1-ssd-mp-0_675.pth` is corrupted - can you try downloading it again?

Thank you for helping me.I tried as what you did in the topic Re-training SSD-Mobilenet.But there are something wrong when I download that like this
ps://nvidia.box.com/shared/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth -O models/mobilenet-v1-ssd-mp-0_675.pth
--2020-09-24 22:56:49-- https://nvidia.box.com/shared/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth
Resolving nvidia.box.com (nvidia.box.com)... 103.116.4.197
Connecting to nvidia.box.com (nvidia.box.com)|103.116.4.197|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: /public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth [following]
--2020-09-24 22:56:50-- https://nvidia.box.com/public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth
Reusing existing connection to nvidia.box.com:443.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://nvidia.app.box.com/public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth [following]
--2020-09-24 22:56:51-- https://nvidia.app.box.com/public/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth
Resolving nvidia.app.box.com (nvidia.app.box.com)... 8.7.198.45, 2001::6ca0:a26d
Connecting to nvidia.app.box.com (nvidia.app.box.com)|8.7.198.45|:443... failed: Connection refused.
Connecting to nvidia.app.box.com (nvidia.app.box.com)|2001::6ca0:a26d|:443... failed: Network is unreachable.

yfc1996, 2020-09-24:
We can down it from https://storage.googleapis.com/models-hao/mobilenet-v1-ssd-mp-0_675.pth

closed 2020-09-24 by yfc1996