dusty-nv/jetson-inference #745: Where can I find the source python codes of jetson.inference and jetson.utils?

davezh, 2020-09-28:
I want to modify the codes of imagenet.py or detectnet.py. I want to customize the object detection.
I want to draw rectangles in the image/video.
Where can I find the source python codes of jetson.inference and jetson.utils?
Where can I find more documents about how to use/modify jetson.inference and jetson.utils python codes?
Thank you.

cognitiveRobot, 2020-09-28:
@davezh start from this page https://github.com/dusty-nv/jetson-inference#hello-ai-world 
One of the best tutorials, very easy to follow.

dusty-nv, 2020-09-28:
The imageNet and detectNet classes are actually implemented in C++ under `jetson-inference/c/` directory.

The Python bindings for these are implemented in `jetson-inference/python/bindings`

davezh, 2020-10-08:
Thank you dusty-nv
I am modifying based on imageNet.py.
Now, I am trying to draw a box in an image where an object is detected, but I got issues.

I tried below codes:
jetson.utils.cudaOverlay(img, img.width, img.height, img_out, img_out.width, img_out.height, img_out.format, 600.0, 200.0)

But I got below error:
Exception: jetson.utils -- cudaOverlay() failed to parse args

Could you please suggest the right way to draw a box in imageNet.py?
Thank you.

dusty-nv, 2020-10-08:
Hi @davezh, see my follow-up to the forum post here:

https://forums.developer.nvidia.com/t/how-to-draw-a-box-in-the-image/156560/3?u=dusty_nv

> Now, I am trying to draw a box in an image where an object is detected, but I got issues.
imageNet doesn't detect individual objects, it classifies the entire image.  Maybe you want to be using detectNet?   detectNet will automatically add the bounding boxes of the detected objects.

Avi-avidan, 2020-10-11:
hi,

dusty-nv, 2020-10-12:
> how can I free memory allocated to a network after successful inference? for example if I want to pass the same image now to a different DNN

Hi @Avi-avidan , you don't need to free the memory to do that.   But what you will want to do, is pass the `overlay='none'` argument to `detectNet.Detect()` so that it doesn't add bounding boxes to your original image.

closed 2023-04-27 by dusty-nv