dusty-nv/jetson-inference #746: Transfer learning cat_dog retraining script throws error.

berkantay, 2020-09-28:
I downloaded the dataset for retraining then next step was to start training script [](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md#re-training-resnet-18-model) but the following error shows up.

```python: relocation error: /home/user2/.local/lib/python2.7/site-packages/torch/lib/libtorch.so: symbol cudnnGetBatchNormalizationForwardTrainingExWorkspaceSize version libcudnn.so.7 not defined in file libcudnn.so.7 with link time reference```

dusty-nv, 2020-09-28:
Hmm...which version of JetPack-L4T are you using?    Are you using the container, or did you install PyTorch manually?

If you installed PyTorch manually, which PyTorch version and wheel did you install?

berkantay, 2020-09-28:
Hey, I use ```NVIDIA Jetson TX2
 L4T 32.1.0 [ JetPack 4.2 ]
   Ubuntu 18.04.2 LTS
   Kernel Version: 4.9.140-tegra
 CUDA 10.0.166
   CUDA Architecture: 6.2
 OpenCV version: 3.3.1
   OpenCV Cuda: NO
 CUDNN: 7.3.1.28
 TensorRT: 5.0.6.3
 Vision Works: 1.6.0.500n
 VPI: NOT_INSTALLED
```
I installed pytorch with the script provided by this repository under ```/home/dasal/jetson-inference/build/install-pytorch.sh```

dusty-nv, 2020-09-28:
I'm not sure why you might have that error.  If possible, your best bet might be to upgrade your device to the latest JetPack, since JetPack 4.2 is getting older at this point.

berkantay, 2020-09-29:
So which version do you recommend 4.4 or 4.3?

berkantay, 2020-09-30:
I upgraded jetpack 4.2 to 4.4 and problem solved but still wondering what was causing that?

dusty-nv, 2020-09-30:
> I upgraded jetpack 4.2 to 4.4 and problem solved but still wondering what was causing that?

I'm not entirely sure, but perhaps that version of PyTorch on JetPack 4.2 was linked against a newer version of cuDNN which removed that symbol.

closed 2020-10-05 by berkantay