dusty-nv/jetson-inference #747: Base model of Object Detectors for transfer learning.

cognitiveRobot, 2020-09-28:
@dusty-nv Thanks a lot for the great repo. I finished re-training `SSD-mobilenet-v1` on my dataset. Now, I would like to retrain `SSD-Mobilenet-v2` and others. I need to compare their accuracy and speed. Where can I get their base models? I want to do transfer learning. Sorry, if I missed, I couldn't locate this info in the docs.

silent-code, 2020-09-28:
Try [here](https://github.com/qfgaohao/pytorch-ssd/blob/master/README.md)

BaosenZ, 2020-10-08:
I use mb2-ssd-lite and pretrained model of mb2-ssd-lite-mp-0_686.pth to try my own data. It always gave me this error:
ValueError: Expected more than 1 value per channel when training, got input size torch.Size([1, 64, 1, 1]).
Do you know how to solve this problem?

cognitiveRobot, 2020-10-23:
Looks like you are feeding wrong inputs.

BaosenZ, 2020-10-23:
@cognitiveRobot Yes, it looks like I feed the wrong inputs. But I did the same thing for mb1-ssd, it works very well. But after I change the command to mb2-lite-ssd to train, it will give me this error. I don't know how to fix that. Can you use mb2-lite-ssd to train your own model? Could you tell me how you do that?

closed 2023-04-27 by dusty-nv