dusty-nv/jetson-inference #748: Anyway you can skip using ONNX files and pass in a UFF file directly?

RoboBoat1, 2020-09-28:
Trying to use pytorch to train a custom model that has already been given to me as a .UFF file.

dusty-nv, 2020-09-28:
Hi @RoboBoat1 , I believe UFF is only used with TensorFlow->TensorRT conversion, and not PyTorch.

RoboBoat1, 2020-09-28:
So @dusty-nv is there no way I can use detectnet to run these models?

RoboBoat1, 2020-09-30:
Something I tried doing was navigating to build/aarch64/bin and I put my .uff and label.txt files into networks, then I tried running:
`./detectnet-camera --model=./networks/unet.uff --class_labels=./networks/Testing_Unet.txt `
since I know detectnet reads in .uff files upon reading some of the other issues people have talked about here. But when I run it, I get these errors

`[TRT]    detected model format - UFF  (extension '.uff')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file ./networks/unet.uff.1.1.6001.GPU.FP16.engine
[TRT]    cache file not found, profiling network model on device GPU
[TRT]    device GPU, loading /home/ubunutu/jetson-inference/build/aarch64/bin/ ./networks/unet.uff
[TRT]    UffParser: Validator error: up_sampling2d_4/ResizeNearestNeighbor: Unsupported operation _ResizeNearestNeighbor
[TRT]    failed to parse UFF model './networks/unet.uff'
[TRT]    device GPU, failed to load ./networks/unet.uff
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model
`
any idea how I could fix this?

closed 2023-04-27 by dusty-nv