dusty-nv/jetson-inference #749: ImageNet not working with trained models on AGX Xavier

Durdle, 2020-09-28:
Hi, my problem is with running ImageNet with trained classification model from my own dataset or the cat_dog dataset. The pretrained models work, but the one I trained doesnt. I looked around and I tried the solution from here: 
https://github.com/dusty-nv/jetson-inference/issues/370
The reinstalling didnt help. Every suggestion is appreciatated.

My setup: Jetson AGX Xavier, Jetpack 4.4, Pytorch 1.6, torchvision (now 0.3.0)

`cd /media/nvidia/xavier_ssd/jetson-inference/python/training/classification
imagenet --model=models/cat_dog/resnet18.onnx --input_blop=input_0 --output_blop=output_0 --labels=data/cat_dog/labels.txt v4l2:///dev/video0`

`[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: Microsoft® LifeCam Studio(TM)
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Microsoft\302\256\ LifeCam\ Studio\(TM\)", v4l2.device.bus_info=(string)usb-3610000.xhci-1, v4l2.device.version=(uint)264588, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 24 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 10/1, 15/2 };
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)960, height=(int)544, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)424, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [9] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [10] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [11] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [12] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [13] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [14] image/jpeg, width=(int)960, height=(int)544, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [15] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [16] image/jpeg, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [17] image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [18] image/jpeg, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [19] image/jpeg, width=(int)432, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [20] image/jpeg, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [21] image/jpeg, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [22] image/jpeg, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] [23] image/jpeg, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 15/1, 10/1, 15/2 };
[gstreamer] gstCamera -- selected device profile:  codec=mjpeg format=unknown width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 ! image/jpeg, width=(int)1280, height=(int)720 ! nvjpegdec ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      mjpeg
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
[... 70 lines cut ...]
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  2
                -- dim #0  1 (SPATIAL)
                -- dim #1  2 (SPATIAL)
[TRT]    
[TRT]    INVALID_ARGUMENT: Cannot find binding of given name: data
[TRT]    failed to find requested input layer data in network
[TRT]    device GPU, failed to create resources for CUDA engine
[TRT]    failed to create TensorRT engine for models/cat_dog/resnet18.onnx, device GPU
[TRT]    failed to load models/cat_dog/resnet18.onnx
[TRT]    imageNet -- failed to initialize.
imagenet:  failed to initialize imageNet
`

If I run the imagenet.py its the same, but at the end it says
`jetson.inference -- imageNet failed to load built-in network 'googlenet'
Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 55, in <module>
    net = jetson.inference.imageNet(opt.network, sys.argv)
Exception: jetson.inference -- imageNet failed to load network
`

dusty-nv, 2020-09-28:
It appears that you have a typo in your command line with `--input_blop=input_0 --output_blop=output_0 `

It should instead be:  `--input_blob=input_0 --output_blob=output_0`

(`_blob` instead of `_blop`)

Durdle, 2020-09-28:
I am so dumb, I read it like a hundred times, still didnt notice. Thank you so much.

closed 2020-09-28 by Durdle