dusty-nv/jetson-inference #750: Custom Pretrained ONNX Model not initializing with ImageNet

jaweiler5454, 2020-09-28:
Hi,

We are attempting to run "imagenet --model=half.onnx v4l2:///dev/video0 output.mp4", where the model file is one that we exported from PyTorch to ONNX. We have run the onnx file through trtexec, and it passes and produces a valid trt model file. However, when we attempt to run ImageNet through the console in the command above, we get the following error. Not sure why we are getting this but any insight is appreciated, or other ways to get more debugging output from the terminal because our error is fairly vague.

Error:

[TRT]  imageNet -- failed to initialize.
imagenet: failed to initialize imageNet

berkantay, 2020-10-05:
I was about the implement same thing with detectnet using my custom trained model in onnx format (with the transfer learning steps provided by this repository).

haohaoqian, 2020-10-27:
Same problem, I'm looking for an update.

dusty-nv, 2020-10-27:
Are you also setting these command line options?

--input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt

i.e. as shown here:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md#training-your-model

haohaoqian, 2020-10-27:
Yes, I run "imagenet.py --model=$NET/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt $DATASET/test/cattail.jpg cattail.jpg", which is mentioned in the docs. And thus I encountered the following errors.
[image]

haohaoqian, 2020-10-27:
> Are you also setting these command line options?
> […](#)
> --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt i.e. as shown here: https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md#training-your-model

Yes, I run "imagenet.py --model=$NET/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt $DATASET/test/cattail.jpg cattail.jpg", which is mentioned in the docs. And thus I encountered the following errors.
[image]

jaweiler5454, 2020-10-27:
Yes, we are running the same command and still getting the same error -- although we are running from the command line rather than python.

dusty-nv, 2020-10-27:
Which version of JetPack are you running?   It looks to be TensorRT 5.0 - if you are on an older JetPack, can you update to JetPack 4.4 or 4.4.1?

On older versions of JetPack, I had a [patched torchvision 0.3 that was installed by the repo](https://github.com/dusty-nv/jetson-inference/blob/7ea5161b59e40f8b05620d53c9bbf7c7cbdff206/tools/install-pytorch.sh#L353) to work around this error, but if you installed torchvision yourself and have older JetPack, you may encounter it.

On recent versions of JetPack, the TensorRT ONNX parser supports the Gather op and doesn't need the patch.

tmh8195, 2020-10-27:
I am also having the same problem and have used all the command line arguments. I am to convert my onnx model using trtexec, but unable to load it using imagenet.

dusty-nv, 2020-10-27:
Thanks - can anyone from this issue confirm if they are using the container, or if not which JetPack version you guys are running?

dusty-nv, 2020-10-27:
For sanity check, I just re-tested this on JetPack 4.4/4.4.1 and was able to run a fresh PyTorch-trained resnet18.onnx model with imagenet program.

It was using the version of PyTorch automatically installed by the jetson-inference repo (PyTorch 1.6 and torchvision 0.7.0).  TensorRT version was 7.1.3.

haohaoqian, 2020-10-28:
Thanks and I'll have a try later！

haohaoqian, 2020-11-06:
I use the latest SD-Image to flash the Nano and it does work!
Thanks for your advice.

closed 2023-04-27 by dusty-nv