dusty-nv/jetson-inference #751: Compiling example detectnet.cpp code with cmake

berkantay, 2020-10-06:
Hello there,

Precompiled version of detectnet works fine and I need to make changes on the source code of the detectnet.cpp . I found that there exist a sample code under the ```~/jetson-inference/examples/detectnet``` called  *detectnet.cpp* and a CMakeList.txt. But when I make shadow build in that directory. 
```mkdir build```
```cd build```
```cmake ..```

**Error : Unknown CMake command "cuda_add_executable".**

What causes this error? How can I fix this?

dusty-nv, 2020-10-06:
Hi @berkantay , you want to still build from the `jetson-inference/build` directory.   The C++ examples like detectnet are part of that larger project, they aren't built standalone like you tried.

berkantay, 2020-10-06:
Hi @dusty-nv , so when it comes to write my own detection code from a live video stream. Which code example should I follow from this repository? I need to store detected object pixels for instance? I should be able to modify that.

dusty-nv, 2020-10-06:
I take it that you are coding in C++ since you were modifying detectnet.cpp - I would just keep with modifying detectnet.cpp.

Note that if you want to extract the object pixels, you will want to call [`detectNet::Detect()`](https://github.com/dusty-nv/jetson-inference/blob/022739abc7631fa853ae63550ec091cd838b2105/c/detectNet.h#L316) with the `OVERLAY_NONE` flag - otherwise it will change the input image with the bounding boxes, labels, ect.

If you still want the overlay, you can manually call [`detectNet::Overlay()`](https://github.com/dusty-nv/jetson-inference/blob/022739abc7631fa853ae63550ec091cd838b2105/c/detectNet.h#L385) after you extract the object pixels.

bhargavravat, 2020-11-04:
hi @dusty-nv : Presently I am also facing the same issue. My use case I want to do inference on the images which I receive from socket connection over the internet. 

Can you suggest me the head way ?

mlao-mesa, 2021-05-07:
If we could compile `detectenet.cpp` from `jetson-inference` like `make exmples` or similar, that would be great. 

Right now it seems that the better option is to use Python. Thanks.

dusty-nv, 2021-05-07:
Hi Michael, if you make changes to detectnet.cpp, you can run `make` and it will re-build the sample.

If you want to make your own detectnet sample outside of the source tree, you can do it like this C++ example:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-example-2.md

hyl-g, 2021-06-07:
Hi there,
I followed the link to generate my own C++ detectnet: 
https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-example-2.md

I got the following error during make:
 fatal error: videoSource.h: No such file or directory
 #include "videoSource.h"

Any idea to solve it?

Thanks

dusty-nv, 2021-06-08:
Hi, try #include <jetson-utils/videoSource.h> instead

https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-example-2.md

hyl-g, 2021-06-08:
Hi dusty,

Thanks for your replay.  It fixed the path for the videoSource.h and videoOutput.h but the compiler still complains the detectNet.h:
 fatal error: detectNet.h: No such file or directory
 #include "detectNet.h"

Anyway to fix that?

Thanks

dusty-nv, 2021-06-08:
Try `#include <jetson-inference/detectNet.h>` instead

hyl-g, 2021-06-08:
Thanks dusty, It works.

closed 2023-04-27 by dusty-nv