dusty-nv/jetson-inference #753: Probels cloning jetson--interference

pjdervan, 2020-10-06:
Hi,

I'm having problems cloning jetson-inference:

desktop:~$ git clone --recursive https://github.com/dusty-nv/jetson-inference
Cloning into 'jetson-inference'...
remote: Enumerating objects: 164, done.
remote: Counting objects: 100% (164/164), done.
remote: Compressing objects: 100% (108/108), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): A TLS packet with unexpected length was received.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

Does anyone have similar problems?

dusty-nv, 2020-10-06:
Hmm I just tried re-cloning and didn't have an issue.

https://www.githubstatus.com/ is not showing a current outage.

pjdervan, 2020-10-06:
OK. Thank you for checking. Must be a problem at mine end.

closed 2020-10-06 by pjdervan

ilackner, 2021-10-31:
Hi I'm getting a similar issue. 
Cloning into 'jetson-inference'...
fatal: unable to access 'https://github.com/dusty-nv/jetson-inference/': Could not resolve host: github.com

dusty-nv, 2021-11-01:
@ilackner can you clone any github repo?   Maybe github was down for a time when you tried before.