dusty-nv/jetson-inference #754: What is the method of calculating FPs in jetson-inference?

jch-wang, 2020-10-10:
Does the method of FPS calculation in "jetson-inference" include image preprocessing, inference, postprocessing, bounding box drawing and image display? Or does it only consider the time required for TensorRT inferencing to calculate FPS?

dusty-nv, 2020-10-10:
Hi jch-wang, if you check the terminal output,  it provides a breakdown for each.  What is displayed in the status bar is the DNN processing time.

jch-wang, 2020-10-13:
I check the terminal output，There will be some information like the following when testing:
```
[TRT]    ------------------------------------------------
[TRT]    Timing Report networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.03965ms  CUDA   0.10342ms
[TRT]    Network       CPU   9.23075ms  CUDA   7.52576ms
[TRT]    Post-Process  CPU   0.01146ms  CUDA   0.01171ms
[TRT]    Total         CPU   9.28186ms  CUDA   7.64090ms
[TRT]    ------------------------------------------------
```
At the same time, the following information will be displayed in the upper left corner of the detection window:
`TensorRT7.1.3|FP16|Network 144FPS`
Is the FPS displayed obtained using the time of (CPU   0.01146ms) + (CUDA   7.52576ms) in the above terminal output ”[TRT]    Network“?

jch-wang, 2020-10-13:
> It is the CUDA time under Network. The CPU / CUDA time aren't added together because they are running at the same time.

hey run at the same time, but the CPU time is longer than CUDA. Why not use a longer CPU time to calculate FPS? This is my new doubt.

dusty-nv, 2020-10-13:
That is because GPU resources are typically the limiting factor in terms of performance, since the networks are run entirely on GPU.   If I was using asynchronous execution with TensorRT, the CPU would not need to wait for GPU - it could be queuing more frames, ect.   That would complicate the program but get you closer to maximum performance.

closed 2020-10-14 by jch-wang