dusty-nv/jetson-inference #755: problem using jeston-inference on RTP stream

mmatesic01, 2020-10-11:
Hello Justin,

I have no issues using RTSP source for object detection with jeton-inference (using latest build).

But I have issues with doing the sam thing with RTP stream (TCP). Is it supporeted?

This is my setup on source side on PI zeroW and this source is reachable with VLC.
_camSet2=' tcpclientsrc host=xxx.xxx.xxx.xxx port=8554 ! gdpdepay ! rtph264depay ! h264parse ! nvv4l2decoder  ! nvvidconv flip-method='+str(flip)+' ! video/x-raw,format=BGRx ! videoconvert ! video/x-raw, width='+str(dispW)+', height='+str(dispH)+',format=BGR ! appsink  drop=true sync=false '_

dusty-nv, 2020-10-12:
Hi @mmatesic01 , I think the issue is that I am using `udpsink` in the GStreamer RTP pipeline - so UDP, not TCP.

Just to confirm, if you change your source's pipeline to UDP, does it work?

BTW, you can try changing it over from UDP to TCP here (this is where the pipeline is constructed for RTP output)

https://github.com/dusty-nv/jetson-utils/blob/916023814a1ac1401380c5d4313454c78c0a5f40/codec/gstDecoder.cpp#L496

mmatesic01, 2020-10-12:
Hi dusty-nv,

I'll try it with your approach and see how it goes.

Thank you.

mmatesic01, 2020-10-15:
Hi,

If I start my stream with this on my RPI:
raspivid -t 0 -w 1296 -h 972 -fps 30 -b 2000000 -awb auto -n  -o - | gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=1 pt=96 ! gdppay ! udpsink host=0.0.0.0 port=5000

How should I reach it on ny Nano with "jetson.utils.videoSource()" + arguments, if any?

Thank you.

mmatesic01, 2020-10-18:
Solved it.

closed 2020-10-18 by mmatesic01