dusty-nv/jetson-inference #758: Building the Project from Source Cmake Issue

ianrheewhi, 2020-10-14:
So I've been working on getting through the setup, but I
I'm building the project from source rather than using the Docker. 
When I try to install PyTorch, I get the following issue: 

`CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_CUDART_LIBRARY (ADVANCED)
    linked by target "jetson-inference" in directory /home/jsh/jetson-inference
    linked by target "imagenet" in directory /home/jsh/jetson-inference/examples/imagenet
    linked by target "imagenet-console" in directory /home/jsh/jetson-inference/examples/imagenet
    linked by target "imagenet-camera" in directory /home/jsh/jetson-inference/examples/imagenet
    linked by target "detectnet-camera" in directory /home/jsh/jetson-inference/examples/detectnet
    linked by target "detectnet" in directory /home/jsh/jetson-inference/examples/detectnet
    linked by target "detectnet-console" in directory /home/jsh/jetson-inference/examples/detectnet
    linked by target "segnet-camera" in directory /home/jsh/jetson-inference/examples/segnet
    linked by target "segnet" in directory /home/jsh/jetson-inference/examples/segnet
    linked by target "segnet-console" in directory /home/jsh/jetson-inference/examples/segnet
    linked by target "camera-capture" in directory /home/jsh/jetson-inference/tools/camera-capture
    linked by target "jetson-utils" in directory /home/jsh/jetson-inference/utils
    linked by target "gl-display-test" in directory /home/jsh/jetson-inference/utils/display/gl-display-test
    linked by target "jetson-utils-python-36" in directory /home/jsh/jetson-inference/utils/python/bindings
    linked by target "jetson-inference-python-36" in directory /home/jsh/jetson-inference/python/bindings
CUDA_TOOLKIT_INCLUDE (ADVANCED)
   used as include directory in directory /home/jsh/jetson-inference
   used as include directory in directory /home/jsh/jetson-inference
   used as include directory in directory /home/jsh/jetson-inference
   used as include directory in directory /home/jsh/jetson-inference
   used as include directory in directory /home/jsh/jetson-inference
   used as include directory in directory /home/jsh/jetson-inference
   used as include directory in directory /home/jsh/jetson-inference
   used as include directory in directory /home/jsh/jetson-inference/docs
   used as include directory in directory /home/jsh/jetson-inference/docs
   used as include directory in directory /home/jsh/jetson-inference/docs
   used as include directory in directory /home/jsh/jetson-inference/docs
   used as include directory in directory /home/jsh/jetson-inference/docs
   used as include directory in directory /home/jsh/jetson-inference/docs
   used as include directory in directory /home/jsh/jetson-inference/examples
   used as include directory in directory /home/jsh/jetson-inference/examples
   used as include directory in directory /home/jsh/jetson-inference/examples
   used as include directory in directory /home/jsh/jetson-inference/examples
   used as include directory in directory /home/jsh/jetson-inference/examples
   used as include directory in directory /home/jsh/jetson-inference/examples
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
   used as include directory in directory /home/jsh/jetson-inference/examples/imagenet
[... 165 lines cut ...]
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings
   used as include directory in directory /home/jsh/jetson-inference/python/bindings

-- Configuring incomplete, errors occurred!
See also "/home/jsh/jetson-inference/build/CMakeFiles/CMakeOutput.log".`

I've deleted the project folder and tried to do it again, but the issue persists. I've followed the instructions to a T,
I  don't know what I'm doing wrong; I've been trying to debug for hours. I would appreciate any and all input.

dusty-nv, 2020-10-21:
Hi @ianrheewhi , can you check that you have the CUDA toolkit installed on your Jetson?

You should be able to find it under `/usr/local/cuda`

If you don't have CUDA, you would need to install it.   I'm not sure which Jetson you are using or version of JetPack, but if you are on JetPack 4.3 or newer you should be able to run `sudo apt-get install nvidia-jetpack`.   Otherwise, you could use SDK Manager tool to install it or if you are on Nano/Xavier-NX you can flash the NVIDIA SD card image.

DennisFaucher, 2021-09-30:
@dusty-nv thanks for this. I just re-images my Xavier with the SDK Manager and nvidia-jetpack was not installed. I will install and see if my cmake errors go away.

closed 2023-04-27 by dusty-nv