dusty-nv/jetson-inference #759: TX2 Video inverted

Awais-commit, 2020-10-14:
Hey, thanks for the wonderful tutorial, I'm new to the Jetson platform and tried your tutorial showed in [https://www.youtube.com/watch?v=bcM5AQSAzUY&feature=youtu.be](https://www.youtube.com/watch?v=bcM5AQSAzUY&feature=youtu.be).

I have tried on my Jetson TX2, it works but the video is upside down. I find some of solutions but didn't work.
I have tried to change the flip method in jetson-inference/utils/camera/gstCamera.cpp but I don't know where to change as .cpp is shown like this [https://github.com/dusty-nv/jetson-utils/blob/916023814a1ac1401380c5d4313454c78c0a5f40/camera/gstCamera.cpp](https://github.com/dusty-nv/jetson-utils/blob/916023814a1ac1401380c5d4313454c78c0a5f40/camera/gstCamera.cpp)  

I read the issue in #Jetson Nano CSI Raspberry Pi Camera V2 upside down 180 degree #571 and tries to find the .cpp file you showed to change here [https://github.com/dusty-nv/jetson-utils/blob/798c416c175d509571859c9290257bd5cce1fd63/camera/gstCamera.cpp#L416](https://github.com/dusty-nv/jetson-utils/blob/798c416c175d509571859c9290257bd5cce1fd63/camera/gstCamera.cpp#L416)

so I replaced it in  jetson-inference/utils/camera/ and run the command $sudo make install but it gives an error and the video was still inverted 180.

my system is Nvidia Jetson TX2 - Jetpack 4.4 [L4T 32.4.3]

I've spent a good amount and couldn't solve the problem. Please guide me on how to solve this

dusty-nv, 2020-10-14:
Hi @Awais-commit , I recommend that you re-clone a fresh copy of the repo (to get the latest updates and undo any changes).   Then you can simply use the `--input-flip=rotate-180` command line argument, for example:

``` bash
$ video-viewer --input-flip=rotate-180 csi://0
```

For more info, please see here:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#command-line-arguments

Awais-commit, 2020-10-15:
Hi @dusty-nv, Thanks for the help. I tried this command `$ video-viewer --input-flip=rotate-180 csi://0` before and now in the command line, This works and show the video correctly. 

But when I run the my-detection.py then the video is upside down. I also tried my-detection.py given in jetson-inference/python/examples/my-detection.py. It also shows the same result (Up-side-down ) 

 By the way, I'm using the code given in (https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-example-2.md) Is there's something to change in the code or what?

Awais-commit, 2020-10-16:
> Can you change this line to the following:
> 
> ```python
> camera = jetson.utils.videoSource("csi://0", argv=["--input-flip=rotate-180"])  
> ```

I tried this but it's still the same

dusty-nv, 2020-10-16:
Hmm ok.  In that case, you can either just run `detectnet.py` (it does basically the same thing, just with additional command-line parsing)

You could also copy the command line parsing from `detectnet.py` to `my-detection.py` and get the flip working again.  But they are essentially the same program aside from the command line parsing.

Awais-commit, 2020-10-19:
> Hmm ok. In that case, you can either just run `detectnet.py` (it does basically the same thing, just with additional command-line parsing)
>
> You could also copy the command line parsing from [...]

I run the `detectnet.py` and it also gives the same output i.e inverted videos. I also tried to copy this command  `input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)` from `detectnet.py` to `my-detection.py` but output is still same.

dusty-nv, 2020-10-19:
Hmm... if you run the `detectnet.py` with the `--debug` flag, do you see something printed during the loading like this:

``` bash
videSource.__init__() argv[1] = '--input-flip=rotate-180'
```

Can you provide your terminal log when you launch the program like `detectnet.py --debug --input-flip=rotate-180`

Awais-commit, 2020-10-20:
Whe I run `detectnet.py --debug` I didn't see `videSource.__init__() argv[1] = '--input-flip=rotate-180'` in terminal log and the output is still inverted.
I also attached the .txt file of the terminal log below:
[detectnet.py --debug.txt](https://github.com/dusty-nv/jetson-inference/files/5409030/detectnet.py.--debug.txt)

However when I run `detectnet.py --debug --input-flip=rotate-180`  the output is not inverted. 
The terminal log file is attached as follow:
[detectnet.py --debug --input-flip=rotate-180.txt](https://github.com/dusty-nv/jetson-inference/files/5409031/detectnet.py.--debug.--input-flip.rotate-180.txt)

dusty-nv, 2020-10-20:
> However when I run `detectnet.py --debug --input-flip=rotate-180` the output is not inverted.
> The terminal log file is attached as follow:
> [detectnet.py --debug --input-flip=rotate-180.txt](https://github.com/dusty-nv/jetson-inference/files/5409031/detectnet.py.--debug.--input-flip.rotate-180.txt)

Ah ok, yes I see in this log the flip method is being set on the camera pipeline.

Does that solve the issue for you using detectnet.py?

Awais-commit, 2020-10-22:
> Ah ok, yes I see in this log the flip method is being set on the camera pipeline.
>
> Does that solve the issue for you using detectnet.py?

Yes, it does solve the issue with `detectnet.py` thanks a lot, but it doesn't work `my-detectnet.py`. What I have to do to solve it with `my-detectnet.py`, do I need to update the code of  `my-detectnet.py`? if yes then where? 
Also where to update the code of `detectnet.py` so that it doesn't require to write `--input-flip.rotate-180` in the terminal every time. Thanks

closed 2021-09-19 by Awais-commit