dusty-nv/jetson-inference #760: Object detection, one box instead of several boxes overlapping?

AK51, 2020-10-14:
Hi,
Great tutorial. I have finished the custom object detection. And I am working on using two servos to track the object.
When an object is detected, it is often to have 2-3 green boxes instead of one. Is there anyway to see just one box?

Thanks.

dusty-nv, 2020-10-21:
Hi @AK51 , you can try decreasing the clustering intersection threshold on this line:

https://github.com/dusty-nv/jetson-inference/blob/501491767ad0127ee2905dcfedace1d7c07b8f2f/c/detectNet.h#L477

If you lower it, it will merge more overlapping bounding boxes.   If you raise it, it will merge less overlapping bounding boxes.

After you change it, run this to recompile:

``` bash
$ cd jetson-inference/build
$ make
$ sudo make install
```

AK51, 2020-10-22:
Yes, it is working. Thx for your help.

closed 2020-10-22 by AK51