dusty-nv/jetson-inference #761: Cannot close GLDisplay in my python application with detectnet

Fkjmp, 2020-10-16:
Hello, 

I created a python application using tkinter and want to use it on the Jetson Nano. When clicking on a button, I want to start the object detection from jetson-inference's detectnet.py. 

My first attempt was to use _os.system('python3 detectnet.py /dev/video0')_ which started the detectnet script from the terminal. So in addition to my tkinter window the window from detectnet.py appeared. Everything worked fine and I was able to just close the detection window after being finished. But as I want to autostart my application on boot-up the os.system line doesn't work because there is no terminal from which to start detectnet. In addition to that I want to make some adjustments in the tkinter application during the detection like changing some labels and also add a signal tone to the detection. 

So I used the code from detectnet.py directly in my own application. Now when clicking the button the detection window opens just like before with os.system and the camera stream and object detection work as well. But when I want to close the detection window so that only my tkinter window remains open, the camera stream freezes but the detection window doesn't close. I also executed this with the --debug flag. The output after _[OpenGL] glDisplay -- the window has been closed_ is exactly the same as the output on using os.system.

I don't have much experience with programming in python or C++ or programming in general. So naturally I have no idea what I could have done wrong.

Any help would be appreciated!

Fkjmp, 2020-10-19:
So today I noticed that the debugging output is _not_ the same. I'm not sure if it is because of some changes I made or if it has been like that before and I just didn't realize.
After the message _[OpenGL] glDisplay -- the window has been closed_ the camera still receives frames and it is not until I close my application completely that the dealloc methods are called (PyTensorNet_Dealloc() ...).
But now my question is if it is somehow possible to call these dealloc methods myself and how/when they are called in the first place (in detectnet.py)?

dusty-nv, 2020-10-19:
> But now my question is if it is somehow possible to call these dealloc methods myself and how/when they are called in the first place (in detectnet.py)?

Hi @Fkjmp , in python you can use the `del` operator to manually dealloc objects, i.e. `del net`.  Normally this is done automatically by python during a garbage collection cycle or when the script terminates.

So my guess is if you do the following, it will close everything down for you:

``` python
del camera
del display
del net
```

Fkjmp, 2020-10-20:
> So my guess is if you do the following, it will close everything down for you:
> 
> ```python
> del camera
> del display
> del net
> ```

Thank you for the help! I tried that and now I'm back to the original question. In debug mode it's the normal output with the dealloc methods but the window is still visible with a frozen picture and I can't close it. (Only when I close my whole application the detectnet window disappears as well.)

Fkjmp, 2020-11-04:
This problem doesn't only appear in my real application but also in a small test application I wrote. Here is my code for the test, maybe it can help to find my mistake:

`# -*- coding: utf-8 -*-

import tkinter as tk
import jetson.inference
import jetson.utils

class Test(tk.Tk):
    
    def __init__(self, *args, **kwargs):
        tk.Tk.__init__(self, *args, **kwargs)
        
        # Set the window's size
        self.geometry('500x350')
        
        # Create the main frame 
        container = tk.Frame(self)
        container.pack(side='top', fill='both', expand=True)
        self.grid_rowconfigure(0, weight=1)
        self.grid_columnconfigure(0, weight=1)
                
        # Create detection button
        btn_det = tk.Button(container, text='Start', width=16, height=4)#, command=self.start)
        btn_det.grid(row=0, column=0, padx=10, pady=10, sticky='NSWE')
    
    def start(self):
        args = ['Test.py', '--debug', '--threshold=0.5', '--network=ssd-mobilenet-v2', '--overlay=box,labels,conf', '--input-codec=raw', '/dev/video0', '']
        opt = ['/dev/video0', '', 'ssd-mobilenet-v2', 'box,labels,conf', 0.5] 
        idx = {'input_URI' : 0, 'output_URI' : 1, 'network' : 2, 'overlay' : 3, 'threshold' : 4}
        
        # Load the object detection network
        net = jetson.inference.detectNet(opt[idx['network']], args, opt[idx['threshold']])
        
        # Connect to the camera device for streaming
        camera = jetson.utils.videoSource(opt[idx['input_URI']], argv=args)
        
        # Create a video output interface 
        display = jetson.utils.videoOutput(opt[idx['output_URI']], argv=args)
        
        while True:
         	# Capture the next image (jetson.utils.cudaImage object))
            img = camera.Capture()
        
         	# Detection network processes the image and returns a list of detections
            detections = net.Detect(img, overlay=opt[idx['overlay']])
            
            print(str(detections) + ' objects detected.\n')
        
            # Visualize the results with OpenGL
            display.Render(img)
                            
            if not display.IsStreaming()or not camera.IsStreaming():
                del net
                del camera
                del display
                break   

if __name__ == '__main__':
    # Function that ends the application
    def closeApp(event):
        app.destroy()
    app = Test()
    
    # Close the application with escape key
    app.bind('<Escape>', closeApp)
    app.mainloop()`

It's the exact same thing: the detection window remains visible (with a frozen picture after trying to close it) until the whole application is closed.

raghuram9581, 2021-02-23:
> This problem doesn't only appear in my real application but also in a small test application I wrote. Here is my code for the test, maybe it can help to find my mistake:
>
> ```
> # -*- coding: utf-8 [...]

Hai, did you find the solution how to close the frozen gldisplay? even i am also facing he same problem. can you help me !

Fkjmp, 2021-02-23:
Hi @raghuram9581, 
I'm sorry, I didn't really find a solution to this problem. But I am using OpenCV now to display the images instead of the jetson.utils.videoOutput. Maybe that's an option for you as well.

Neltherion, 2021-11-01:
I'm also having this problem.

closed 2023-04-27 by dusty-nv