dusty-nv/jetson-inference #763: I have a question during the tutorial.

ingbeeedd, 2020-10-20:
I am currently viewing the image recognition tutorial. I bought it because Recommended System Requirements said I needed training PCs, but do I have to run all the tutorials I'm looking at on the Jetson board? Is the docker installed on the jetson? I was wondering if the PC I bought from the training in the back was used. ? thank you for your regard

dusty-nv, 2020-10-21:
Hi @ingbeeedd , the Recommended System Requirements don't list a PC as required:   https://github.com/dusty-nv/jetson-inference#recommended-system-requirements

You can do the training onboard a Jetson.   The Docker container is run on Jetson.

However, if you do happen to have a Linux PC with NVIDIA GPU card in it, you can train a lot faster on your PC than you would on Jetson.   Training on Jetson is mostly for educational/illustrative purposes.   If you have a large dataset/model, you would most likely want to train it on a PC or in the cloud.

ingbeeedd, 2020-10-22:
Thank you for your answer. The last question is, is there any problem with the docker image and container that works at Jetson? In other words, is it possible to use the same docker environment even if hardware changes? Thank you for your regards.

closed 2020-11-02 by ingbeeedd

ingbeeedd, 2020-12-17:
The Docker eventually relies on the Linux kernel.