dusty-nv/jetson-inference #764: imagenet sample run much slower use the engine cache file

sky-fun, 2020-10-21:
Hi,
I build this project in my NVIDIA Jetson Xavier NX device,when i run the imagenet sample with 
`./imagenet --network=resnet-18 images/jellyfish.jpg images/test/output_jellyfish.jpg` for the first time, the cost time is about 4ms
```
[TRT]    ------------------------------------------------
[TRT]    Timing Report networks/ResNet-18/ResNet-18.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.08611ms  CUDA   0.04842ms
[TRT]    Network       CPU   4.09130ms  CUDA   3.91024ms
[TRT]    Post-Process  CPU   0.11168ms  CUDA   0.11120ms
[TRT]    Total         CPU   4.28910ms  CUDA   4.06986ms
[TRT]    ------------------------------------------------
```

but after that each time i run the same command ,the total cost time is over 22ms

```
[TRT]    ------------------------------------------------
[TRT]    Timing Report networks/ResNet-18/ResNet-18.caffemodel
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.09654ms  CUDA   0.19050ms
[TRT]    Network       CPU  24.41387ms  CUDA  23.46602ms
[TRT]    Post-Process  CPU   0.10746ms  CUDA   0.10650ms
[TRT]    Total         CPU  24.61787ms  CUDA  23.76301ms
[TRT]    ------------------------------------------------
```

after i read the code, i found that difference between the first time and the latter ones : the sample will generate a engine cache file `networks/ResNet-18/ResNet-18.caffemodel.1.1.7100.GPU.FP16.engine`, and after that the sample will load the .engine file instead of build engine from caffemodel;
when i delete the `networks/ResNet-18/ResNet-18.caffemodel.1.1.7100.GPU.FP16.engine` file, and run the command again, the cost time comes back to 4ms.
so i concerned that which cost time is the real one of the resnet-18 sample and why this appended?

environment of the  NVIDIA Jetson Xavier NX  detected by [jetsonUtilities](https://github.com/jetsonhacks/jetsonUtilities) is
```
NVIDIA Jetson Xavier NX (Developer Kit Version)
 L4T 32.4.2 [ JetPack 4.4 DP ]
   Ubuntu 18.04.4 LTS
   Kernel Version: 4.9.140-tegra
 CUDA 10.2.89
   CUDA Architecture: 7.2
 OpenCV version: 4.1.1
   OpenCV Cuda: NO
 CUDNN: 8.0.0.145
 TensorRT: 7.1.0.16
 Vision Works: 1.6.0.501
 VPI: 0.3.7
```

dusty-nv, 2020-10-21:
Hi @sky-fun, I think it has to do with the dynamic clock rates.  On the first run when the TensorRT engine is created, the CPU/GPU clocks are already spun up, because TensorRT was running optimizations.   But on the second run when the TensorRT engine is simply loaded from disk, the CPU/GPU clocks are still low or idle.

You can try running `sudo jetson_clocks` beforehand, and the processor clocks will stay at their maximums.

Also processing a single image one time isn't great way to measure sustained performance.   Instead, I recommend you run it on a camera feed or a video file to get the realtime performance.

sky-fun, 2020-10-22:
@dusty-nv , thanks, after i run `sudo jetson_clocks`, the cost times of using caffemodel and engine file are barely the same;
i will close this issue.

Ps.
By saying that i need to `run it on a camera feed or a video file get the realtime performance`,do you meaning decode the video will affect the performance， or performance of different images can be very different？

closed 2020-10-22 by sky-fun