dusty-nv/jetson-inference #765: Update to Ubuntu 20.04

Buki94, 2020-10-22:
Hey guys,

first i am satisfied with this system and til today it works very well. 

Now I updated my Ubuntu-os from 18.04 to 20.04 and as the System Requirements says, the Jetson Nano with yolo dosent work any more. 

Are there any plans to run yolo on the jetson nano with ubuntu 20.04 or should I downgrade the OS to 18.04?

Thanks in advance!

dusty-nv, 2020-10-27:
Hi @Buki94 , if you are having problems I would downgrade back to 18.04.   The official JetPack image is still on 18.04.

Zumbalamambo, 2021-05-17:
Is it possible to install it in Ubuntu 20.0.4 desktop?

dusty-nv, 2021-05-17:
Hi @Zumbalamambo, do you mean YOLO on Ubuntu 20.04 on x86 desktop?   I'm not sure which YOLO is being referred to because YOLO isn't in this repo.

g30ba1, 2021-06-10:
> Hi @Buki94 , if you are having problems I would downgrade back to 18.04. The official JetPack image is still on 18.04.

Hi there 😀

I've made the BIG mistake to upgrade my Jetson Nano (JetPack 4.5.1 with Ubuntu 18.04) using software updater app to Ubuntu 20.04

Now, all kind of programs and settings are BROKE 😥

Is there a way to downgrade my installation?

Zumbalamambo, 2021-06-11:
> Hi there 😀
>
>
>
> I've made the BIG mistake to upgrade my Jetson Nano (JetPack 4.5.1 with Ubuntu 18.04) using software updater app to Ubuntu 20.04
>
>
>
> Now, all kind of programs and settings are BROKE [...]

Don't use deep stream or ssd model that is being offered in this repo. It was so inaccurate when you train it on custom dataset. Better train with yolov4, export it as onnx and infer using onnx runtime-gpu. You can then develop on your desktop having u Ubuntu 20.0.4 and deploy the same code in jetson without any change. It will give a better accuracy and performance without having to undergo all these unwanted conversions. I wasted my time on TLT as well as on training SSD given in this repo. It just won't work.

closed 2023-04-27 by dusty-nv