dusty-nv/jetson-inference #766: Retraining SSD Mobilenet - Downloading Images Error

irfwas, 2020-10-22:
Hello, after following all the steps to retrain the SSD-Mobile net. When it comes to the 'Limiting the Amount of Data' section. By simply copying and pasting your code, I get this error:

`$ python3 open_images_downloader.py --stats-only --class-names "Apple,Orange,Banana,Strawberry,Grape,Pear,Pineapple,Watermelon" --data=data/fruit `

Traceback (most recent call last):
  File "open_images_downloader.py", line 2, in <module>
    import boto3
  File "/home/irfwas/.local/lib/python3.6/site-packages/boto3/__init__.py", line 16, in <module>
    from boto3.session import Session
  File "/home/irfwas/.local/lib/python3.6/site-packages/boto3/session.py", line 17, in <module>
    import botocore.session
  File "/home/irfwas/.local/lib/python3.6/site-packages/botocore/session.py", line 30, in <module>
    import botocore.credentials
  File "/home/irfwas/.local/lib/python3.6/site-packages/botocore/credentials.py", line 34, in <module>
    from botocore.config import Config
  File "/home/irfwas/.local/lib/python3.6/site-packages/botocore/config.py", line 16, in <module>
    from botocore.endpoint import DEFAULT_TIMEOUT, MAX_POOL_CONNECTIONS
  File "/home/irfwas/.local/lib/python3.6/site-packages/botocore/endpoint.py", line 22, in <module>
    from botocore.awsrequest import create_request_object
  File "/home/irfwas/.local/lib/python3.6/site-packages/botocore/awsrequest.py", line 25, in <module>
    import botocore.utils
  File "/home/irfwas/.local/lib/python3.6/site-packages/botocore/utils.py", line 30, in <module>
    from urllib3.util.url import IPV6_ADDRZ_RE
ImportError: cannot import name 'IPV6_ADDRZ_RE'

dusty-nv, 2020-10-27:
Hi @irfwas , I wonder if this is related to this recent issue in botocore package?

https://github.com/boto/botocore/issues/2187

irfwas, 2020-10-30:
Hello @dusty-nv, I managed to fix it as I did not preinstall the botocore package. Thank you.

closed 2020-10-30 by irfwas