dusty-nv/jetson-inference #767: Document for jetson.utils and jetson.inference

magallardo, 2020-10-22:
Hello,
Sorry but I am just a beginner and was looking for some docs for the jetson.utils and jetson.inference packages.

I have followed the tutorial and everything is working fine. However, I have created a Flask wrapper to my custom python project to receive inference requests through a REST interface.
When doing that, I am receiving an image encoded in base64 format.  I need to know the type of image format the imagenet classify operation requires.  I have decoded the image I receive and tried passing  that, but I get the following error:
class_idx, confidence = net.Classify(img)
Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object

I need help converting my base64 encoded image to one of the formats expected by the classify operation.  

Any help will be greatly appreciated.

dusty-nv, 2020-10-22:
Hi @magallardo , this document might help you:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#image-capsules-in-python

imageNet/detectNet/segNet expects to receive `cudaImage` object in one of these formats:

* `rgb8`
* `rgba8`
* `rgb32f`
* `rgba32f`

If you convert your data from base64 into a numpy array, you can use `cudaFromNumpy()` function to get the `cudaImage` object:

https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-from-numpy.py

magallardo, 2020-10-22:
@dusty-nv Thanks for your suggestion and I was able to get it working.  I don't know if it is the most efficient way, but it is working!!
Following is what I did and used Image from pillow.

b64decoded = base64.b64decode(image)
image = Image.open(io.BytesIO(b64decoded))
image_np = np.array(image)
cuda_img = jetson.utils.cudaFromNumpy(image_np)

Thanks again.

closed 2023-04-27 by dusty-nv

QuocDuyTran, 2023-05-22:
import jetson.inference
import jetson.utils
import cv2
import numpy as np

net = None
camera = None

def initialize_detector():
	global net, camera
	net = jetson.inference.detectNet("ssd-mobilenet-v2")
	camera = jetson.utils.videoSource("/dev/video0")      # '/dev/video0' for V4L2

def get_detections():
	person_detections = []
	img = camera.Capture()
	detections = net.Detect(img)
	for detection in detections:
		if detection.ClassID == 1: #remove unwanted classes
			person_detections.append(detection)
	fps = net.GetNetworkFPS()

	return person_detections, fps, jetson.utils.cudaToNumpy(img)

import detector_mobilenet as detector

initialize_detector()

while True:
    detections, fps, image = get_detections()
    cv2.putText(image, str(fps), (50, 50), cv2.FONT_HERSHEY_SIMPLEX , 1, (0, 0, 255), 3, cv2.LINE_AA) 
    cv2.imshow("out", image)
    cv2.waitKey(1)
    
    I have bug with my code about " detections = net.Detect(img)" 
    Exception: jetson.utils -- function wasn't passed a valid cudaImage or cudaMemory object"

dusty-nv, 2023-05-22:
@QuocDuyTran, try adding this check after `img = camera.Capture()`:

```
if img is None:
   return
```

camera.Capture() can return None in the event of a capture timeout, so your `get_detections()` function should check for it and try again

SJavad, 2023-07-10:
I trying to import the jetson_utils.cudaFromNumpy and doesn't work
it says to me:
ModuleNotFoundError: No Module named 'jetson_utils.cudaFromNumpy'
I try to rebuild the jetson_inference repository
it builds successfully.
but it doesn't work... 
how can I solve that?

dusty-nv, 2023-07-10:
@SJavad are you able to run other scripts that use jetson_inference/jetson_utils (like video-viewer.py, imagenet.py, or detectnet.py) or is it just that cudaFromNumpy() function that gives you the error?

dusty-nv, 2023-07-10:
When I run `from jetson_utils import cudaFromNumpy`, I'm able to import it ok.

I wonder if when you build jetson-inference, it didn't detect the numpy dev packages?

SJavad, 2023-07-10:
> When I run `from jetson_utils import cudaFromNumpy`, I'm able to import it ok.
> 
> I wonder if when you build jetson-inference, it didn't detect the numpy dev packages?

I build the Jetson-inference successfully and all numpy packages like python-numpy python3-numpy are installed

SJavad, 2023-07-11:
> When I run `from jetson_utils import cudaFromNumpy`, I'm able to import it ok.
> 
> I wonder if when you build jetson-inference, it didn't detect the numpy dev packages?

@dusty-nv 
when I import the: from jetson_utils import cudaFromNumpy with python 2.7
it is import successfully 
but my problem for python3
looks like the jetson_utils it is not build for python3...
how can I fix that ?

dusty-nv, 2023-07-11:
> -- detecting Python 3.6...
> -- found Python version:  3.6 (3.6.9)
> -- found Python include:  /usr/include/python3.6m
> -- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
> -- CMake module path:  /home/deep/SSD_TEST_PYLON/jetson-utils/cuda;/home/deep/SSD_TEST_PYLON/jetson-utils/python/bindings
> -- NumPy not found

Can you try installing `python3-numpy` package from apt (although jetson-inference should have already done this), or `pip3 install --upgrade numpy`

SJavad, 2023-07-12:
> Can you try installing `python3-numpy` package from apt (although jetson-inference should have already done this), or `pip3 install --upgrade numpy`

@dusty-nv 
yes I try to install `python3-numpy` and also upgrade `pip3 install --upgrade numpy` but two commands say the numpy version is the latest version and installed
also when I try to import the jetson_utils it gives me a warning:
```cpp
jetson.utils -- compiled without Numpy array conversion support (warning)
jetson.utils -- if you wish to have support for converting NumPy arrays.
jetson.utils -- first run 'sudo apt-get install python-numpy python3-numpy'
warning: importing jetson.utils is deprecated. please 'import jetson_utils' instead
```
also when I try to import the jetson_utils (no jetson.utils), can't find it at all. even cudaMemcpy...
is there any environment variables that set the numpy library address? 

my problem is that I can't import cudaFromNumpy...

SJavad, 2023-07-12:
> > > -- detecting Python 3.6...
> > > -- found Python version:  3.6 (3.6.9)
> > > -- found Python include:  /usr/include/python3.6m
> > > -- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
> > > -- CMake module path:  /home/deep/SSD_TEST_PYLON/jetson-utils/cuda;/home/deep/SSD_TEST_PYLON/jetson-utils/python/bindings
> > > -- NumPy not found
> > 
> > 
> > Can you try installing `python3-numpy` package from apt (although jetson-inference should have already done this), or `pip3 install --upgrade numpy`
> 
> @dusty-nv yes I try to install `python3-numpy` and also upgrade `pip3 install --upgrade numpy` but two commands say the numpy version is the latest version and installed also when I try to import the jetson_utils it gives me a warning:
> 
> ```c++
> jetson.utils -- compiled without Numpy array conversion support (warning)
> jetson.utils -- if you wish to have support for converting NumPy arrays.
> jetson.utils -- first run 'sudo apt-get install python-numpy python3-numpy'
> warning: importing jetson.utils is deprecated. please 'import jetson_utils' instead
> ```
> 
> also when I try to import the jetson_utils (no jetson.utils), can't find it at all. even cudaMemcpy... is there any environment variables that set the numpy library address?
> 
> my problem is that I can't import cudaFromNumpy...

also, try to build the numpy from the source.
but nothing happens.

dusty-nv, 2023-07-12:
> ```c++
> jetson.utils -- compiled without Numpy array conversion support (warning)
> jetson.utils -- if you wish to have support for converting NumPy arrays.
> jetson.utils -- first run 'sudo apt-get [...]

@SJavad I'm not sure why it's not detecting numpy for python3 on your system - you could try adding debug print-outs to this file to see what's happening:

https://github.com/dusty-nv/jetson-utils/blob/f0bff5c502f9ac6b10aa2912f1324797df94bc2d/python/bindings/FindNumPy.cmake

or you can run the jetson-inference container, which has this all built-in.

SJavad, 2023-07-13:
@dusty-nv 
I apologize for this long conversation, thank you for all your efforts ❤🌸