dusty-nv/jetson-inference #768: Detectron2 Integration

berkantay, 2020-10-23:
Hello Dustin,

I was completed your tutorial on inference with jetson. It is really nice project. I ran my own detection with my custom dataset (drone images) using transfer learning based on Mobilenet SSD. But according to my research on papers [](https://www.researchgate.net/publication/337537833_Drone_Detection_in_Long-Range_Surveillance_Videos) and [](https://sci-hub.do/10.1109/AVSS.2017.8078464) I decide to compare mobilenet ssd vs faster rcnn resnet101. I will use same dataset for the comparison.

I have used detectron2 for transfer learning training based on faster rcnn resnet101 with same dataset but in coco format. Now I have my .pth (pytorch i guess) model. How can I use this model with ```detectnet``` code?

Note: I am open to any other suggestions for training?

cognitiveRobot, 2020-10-23:
I am also interested in this workflow, i.e. train using `detectron2` and inference using `jetson-inference` on jetson.

dusty-nv, 2020-10-23:
I'm not sure if detectron2 is supported in TensorRT.   I believe TensorFlow version of maskRCNN is, because there is a maskRCNN sample found at `/usr/src/tensorrt/samples/sampleUffMaskRCNN` in recent version of JetPack.  I am not sure of the performance, e.g. on Jetson Nano.

Supporting it in the library would likely require some significant code additions to the pre/post-processing of either detectNet and/or segNet class.  I'm not terribly familiar with maskRCNN, it may require it's own hybrid object since it is both detectNet _and_ segNet.

Anyways, before working on it myself, I would prefer if PyTorch-trained ONNX version of maskRCNN was working in TensorRT, and that it was realtime performance on Jetson.

berkantay, 2020-10-23:
I am new to these topics but I think detectron2 is able to convert trained model to ```.onnx``` format AFAIK (https://detectron2.readthedocs.io/modules/export.html#detectron2.export.export_onnx_model). Documents of TensorRT also mentions about .onnx (https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html#overview). So is it a valid solution? Am I on wrong side?

dusty-nv, 2020-10-23:
It depends on if TensorRT can load/run the ONNX model that is exported (if all the layers/ops are supported, ect).   From the detectron2 documentation that you linked to, this unfortunately seems unlikely as they are using custom ops:

`Export the model to ONNX format. Note that the exported model contains custom ops only available in caffe2, therefore it cannot be directly executed by other runtime.`

berkantay, 2020-10-23:
So it does mean that I cannot use TensorRT engine. Well, does it also mean that there is no way that I can use exported .onnx model?

dusty-nv, 2020-10-23:
> Well, does it also mean that there is no way that I can use exported .onnx model?

Yea, it sounds like you can only use the exported ONNX model in caffe2 because of the custom op.   I suppose you could write custom op parsers/executors to support it in other frameworks, but it would probably take some significant work (if it works at all)

berkantay, 2020-10-23:
Okay thank you so much @dusty-nv. Probably It seems like I will implement my training algorithm and network on tensorflow than use tensorrt. I do not have that much time :)

valli28, 2020-11-18:
@berkantay What did you end up doing?
I am also interested in using detectron2-trained models on the Jetson Nano or other Jetson devices, and I am curious as to how you've continued with your project :)

berkantay, 2020-11-18:
Hello @valli28 I changed my path to yolov5 that has already onnx export and tensorrt deployment tutorials. Currently my work on detectron2 is frozen. But I am having troubles with onnx format exported from yolov5. #800

valli28, 2020-11-18:
Thank you @berkantay. I think my best course of action is to train a new model using segnet on my dataset. It's not huge, so I think that's the best solution for me right now. 
I hope FAIR or Nvidia increases compatibility between detectron2 and TensorRT in the near future.

berkantay, 2020-11-18:
Hope they will @valli28

closed 2023-04-27 by dusty-nv