dusty-nv/jetson-inference #776: How to make a Directory to train SSD-based Object Detection in PyTorch

saifurrehman4114, 2020-10-27:
Hi, I have trained the model using the Camera Capture tool it was brilliant.

Now I am willing to train this model using my own data set which has about 100 Pictures of me not annotated yet,
1. I am unable to figure first how to annotate them?
2. How to make a directory structure as Camera Capture makes automatically to run this model?

If you could make a video it would be really helpful.

dusty-nv, 2020-10-27:
Hi @saifurrehman4114 , I recommend to use the CVAT tool - https://github.com/openvinotoolkit/cvat

I tried it and it worked - I needed to make a labels.txt in my dataset, but otherwise it exported the data in Pascal-VOC structure.

I also needed to make a few minor tweaks today to the training script for it, so you should update or re-clone the repo.  And if you are using the containers, run the `docker/pull.sh` script to get the latest container.

saifurrehman4114, 2020-10-28:
Thanks  @dusty-nv. 
1.Have you Tried LabelIMG tool to prepare data set?
2. After using CVAT did you made JPEG folder to store images in them & train?

dusty-nv, 2020-10-28:
> Have you Tried LabelIMG tool to prepare data set?

Initially I tried LabelImg, but it did not create the proper Pascal-VOC directory structure - just the annotations.

> 2. After using CVAT did you made JPEG folder to store images in them & train?

When you export your dataset from CVAT in Pascal VOC format, you will download a zip file that already has the JPEG folder, annotations folder, imagesets folder, ect.   All you need to do is create a labels.txt for it, and put it inside your extracted dataset.

saifurrehman4114, 2020-10-29:
Thanks @dusty-nv. In CVAT when someone is the task Dashboard & someone tries to export Data Set in Pascal Format JPEG Folder doesn't come In my case working on their web application.

dusty-nv, 2020-10-29:
Hmm, I notice that sometime, it can take a couple minutes for CVAT to prepare the zipped Pascal-VOC file for you to download (maybe longer if your dataset is long)

You can also try clicking on `Tasks` menu at top.   Then `Open` your dataset, and try to export again from the `Actions` menu.

I believe you can also install CVAT tool locally on Linux/Windows/Mac.

saifurrehman4114, 2020-10-30:
Thanks @dusty-nv for the Guidance. Looking on creating something great.

closed 2020-10-31 by saifurrehman4114