dusty-nv/jetson-inference #777: ERROR WHEN RUNNING CUSTOMISED DETECTION

spy-maker, 2020-10-28:
I am getting errors when running .py file for own customised detections. I created labels and captured images of different types of lego toys for my own customised detection. I didn't face any issue while training the model but when I tried to run the developed model I couldn't get the results.

This is my terminal screen after I run the program:

jetbot@jetbot-desktop:~/jetson-inference$ python3 deeplearning1.py 
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 4 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: PowerService: requested_clock_Hz=2205000
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
jetson.inference -- imageNet loading network using argv command line params

imageNet -- loading classification network model from:
         -- prototxt     (null)
         -- model        /home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx
         -- class_labels /home/jetbot/jetson-inference/myTrain/labels.txt
         -- input_blob   'input_0'
         -- output_blob  'output_0'
         -- batch_size   1

[TRT]    TensorRT version 6.0.1
[TRT]    loading NVIDIA plugins...
[TRT]    Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]    Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]    Plugin Creator registration succeeded - NMS_TRT
[TRT]    Plugin Creator registration succeeded - Reorg_TRT
[TRT]    Plugin Creator registration succeeded - Region_TRT
[TRT]    Plugin Creator registration succeeded - Clip_TRT
[TRT]    Plugin Creator registration succeeded - LReLU_TRT
[TRT]    Plugin Creator registration succeeded - PriorBox_TRT
[TRT]    Plugin Creator registration succeeded - Normalize_TRT
[TRT]    Plugin Creator registration succeeded - RPROI_TRT
[TRT]    Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]    Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]    detected model format - ONNX  (extension '.onnx')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file /home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx.1.1.6001.GPU.FP16.engine
[TRT]    cache file not found, profiling network model on device GPU
[TRT]    device GPU, loading /usr/bin/ /home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx
----------------------------------------------------------------
Input filename:   /home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx
ONNX IR version:  0.0.3
Opset version:    9
Producer name:    pytorch
Producer version: 0.4
Domain:           
Model version:    0
Doc string:       
[... 139 lines cut ...]
  name: "axis"
  i: 0
  type: INT
}
doc_string: "/usr/local/lib/python3.6/dist-packages/torchvision/models/resnet.py(161): forward\n/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py(477): _slow_forward\n/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py(487): __call__\n/usr/local/lib/python3.6/dist-packages/torch/nn/modules/container.py(92): forward\n/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py(477): _slow_forward\n/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py(487): __call__\n/usr/local/lib/python3.6/dist-packages/torch/jit/__init__.py(252): forward\n/usr/local/lib/python3.6/dist-packages/torch/nn/modules/module.py(489): __call__\n/usr/local/lib/python3.6/dist-packages/torch/jit/__init__.py(197): get_trace_graph\n/usr/local/lib/python3.6/dist-packages/torch/onnx/utils.py(192): _trace_and_get_graph_from_model\n/usr/local/lib/python3.6/dist-packages/torch/onnx/utils.py(224): _model_to_graph\n/usr/local/lib/python3.6/dist-packages/torch/onnx/utils.py(281): _export\n/usr/local/lib/python3.6/dist-packages/torch/onnx/utils.py(104): export\n/usr/local/lib/python3.6/dist-packages/torch/onnx/__init__.py(27): export\nonnx_export.py(77): <module>\n"

--- End node ---
ERROR: onnx2trt_utils.hpp:347 In function convert_axis:
[8] Assertion failed: axis >= 0 && axis < nbDims
[TRT]    failed to parse ONNX model '/home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx'
[TRT]    device GPU, failed to load /home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx
[TRT]    failed to load /home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx
[TRT]    imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'alexnet'
Traceback (most recent call last):
  File "deeplearning1.py", line 16, in <module>
    net=jetson.inference.imageNet('alexnet',['--model=/home/jetbot/jetson-inference/python/training/classification/myModel/resnet18.onnx','--input_blob=input_0','--output_blob=output_0','--labels=/home/jetbot/jetson-inference/myTrain/labels.txt'])
Exception: jetson.inference -- imageNet failed to load network
GST_ARGUS: Cleaning up

(python3:31757): GStreamer-CRITICAL **: 13:44:14.401: gst_mini_object_set_qdata: assertion 'object != NULL' failed
CONSUMER: Done Success
GST_ARGUS: Done Success
GST_ARGUS: 
PowerServiceHwVic::cleanupResources

dusty-nv, 2020-10-28:
Hi @spy-maker , which version of JetPack are you using?     Can you try upgrading to JetPack 4.4 or 4.4.1?      Those JetPack version include TensorRT 7, and the error should go away.

spy-maker, 2020-10-30:
> Hi @spy-maker , which version of JetPack are you using?     Can you try upgrading to JetPack 4.4 or 4.4.1?      Those JetPack version include TensorRT 7, and the error should go away.

Currently, I am using 4.3 jetpack. If I upgrade it to the 4.4 or 4.4.1 will it have any impact on my previous program files or do I need to download using new sd card?

spy-maker, 2020-11-03:
Hi,
Is it possible to update or upgrade jetpack version from 4.3 to 4.4 or 4.4.1 ? Without reflashing the new image ?

closed 2021-03-15 by spy-maker