dusty-nv/jetson-inference #778: Processing Images with TensorRT: listDir('/usr/bin/=models/fruit/labels.txt') - found no matches

chrisTopp84, 2020-10-28:
Hi Dusty,

I'm trying to run this part -> "Processing Images with TensorRT" of the repository.

This is my input:
`nvidia@nvidia-desktop:~/jetson-inference/python/training/detection/ssd$ detectnet.py --model=models/fruit/ssd-mobilenet.onnx --labels =models/fruit/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "data/images/fruit_*" data/fruit/test/fruit_%i.jpg`

And this is the error:
```
listDir('/usr/bin/=models/fruit/labels.txt') - found no matches
[image] imageLoader -- failed to find '=models/fruit/labels.txt'
Traceback (most recent call last):
  File "/usr/local/bin/detectnet.py", line 54, in <module>
    input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device
```
Maybe i just made a mistake with the path i entered. But i don't know.
Can you help me with that? Or maybe somebody else?

If you need more informatione please let me know.

BR chris

dusty-nv, 2020-10-28:
Hi @chrisTopp84 , I think the issue in your original command line is that there is a space between `--labels` and `=models/fruit/labels.txt`:    `--labels =models/fruit/labels.txt`

Remove that space and I think it should run then:

``` bash
detectnet.py --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "data/images/fruit_*" data/fruit/test/fruit_%i.jpg
```

chrisTopp84, 2020-10-28:
Thanks a lot! It is running now :)

closed 2023-04-27 by dusty-nv