dusty-nv/jetson-inference #781: Retraining SSD Mobilenet Issue -Detectnet failed to initialize

irfwas, 2020-10-30:
Hello @dusty-nv,

I have two issues:

1. Trying to train for 100 epochs, the system shows an error due to low memorry and force quits at ~50 epochs. I tried to resume but there is no clear information on the code on how to resume from this issue. Could you kindly provide me the code or steps to resume back at the 50 epoch mark?

2. I converted my file to an onxx file and tried to run the file but i got the error shown below. Now, this error occurs for any detection dataset whether its SSD Mobilenet V2, SSD Inception V2 or others, although it worked before I tried to retrain the network. 

**:~/jetson-inference/python/training/detection/ssd$ detectnet.py --model=models/adas/sss-mobilenet.onnx --labels=models/adas/labels.txt --input=blob=input_0 --output-cvg=scores --output-bbox=boxes csi://0**

jetson.inference -- detectNet loading network using argv command line params

detectNet -- loading detection network model from:
          -- prototxt     NULL
          -- model        models/adas/sss-mobilenet.onnx
          -- input_blob   'data'
          -- output_cvg   'scores'
          -- output_bbox  'boxes'
          -- mean_pixel   0.000000
          -- mean_binary  NULL
          -- class_labels models/adas/labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]    TensorRT version 7.1.3
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - ONNX  (extension '.onnx')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file .1.1.7103.GPU.FP16.engine
[TRT]    cache file not found, profiling network model on device GPU

error:  model file 'models/adas/sss-mobilenet.onnx' was not found.
        if loading a built-in model, maybe it wasn't downloaded before.

        Run the Model Downloader tool again and select it for download:

           $ cd <jetson-inference>/tools
           $ ./download-models.sh

[TRT]    detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load network
Traceback (most recent call last):
  File "/usr/local/bin/detectnet.py", line 51, in <module>
    net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)
Exception: jetson.inference -- detectNet failed to load network

dusty-nv, 2020-11-04:
Here is a link to the forum post discussing this issue:   https://forums.developer.nvidia.com/t/how-to-utilize-class-id-in-object-detection-to-control-gpio-pins/158433/4?u=dusty_nv

dusty-nv, 2022-03-15:
> error: model file 'models/adas/sss-mobilenet.onnx' was not found.

It cannot find the path to your model.  Are you sure you don't mean `ssd-mobilenet.onnx` instead of `sss-mobilenet.onnx`?

ihebchh, 2022-03-18:
Thank you for help it works correctly now.
I will use an ip camera for this application but the streaming is not in
real time  there is a late of more than  2 seconds.
I'm using this command:
detecnet --model=models/person/ssd-mobilenet.onnx
--labels=models/person/labels.txt --input_blob=input_0 --output-cvg=scores
--output-bbox=boxes ***@***.***:554
Please help
Thanks

closed 2023-04-27 by dusty-nv