dusty-nv/jetson-inference #782: Detecting defects on metallic surfaces

Fkjmp, 2020-10-30:
Hello,

as the title suggests I am trying to detect defects on metallic surfaces with jetson-inference on the Jetson Nano. 
To get used to the training I created a test dataset to detect black and red pieces of paper. I retrained SSD-Mobilenet as described in Hello AI World with my black and red dataset and it worked out well.

Afterwards I started retraining with datasets of surface defects. But because after some time I still did not see any results or much progress I gave that up for now and started trying to detect at least one specific defect. But I did not see any progress with that eather. It is always the same: When I set the detection threshold in detectnet.py with my own model to 0.2 defects are deteted all over the surface (where there are no defects). But with 0.3 or higher, nothing is detected at all. 

So now I am wondering if I am doing something wrong or if I have to make some changes for that kind of objects to detect. Or do I just need to add even more data?

take5553, 2021-07-20:
I had the same issue. I left jetson nano (2GB) working during a whole night but it was still stucked. It was my first time and the second try worked properly. So, making many tries is the only solution?

Here is what I got at the first time.
~~~
File "train_ssd.py", line 346, in <module>
val_loss, val_regression_loss, val_classification_loss = test(val_loader, net, criterion, DEVICE)
File "train_ssd.py", line 150, in test
for _, data in enumerate(loader):
File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 363, in __next__
data = self._next_data()
File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 974, in _next_data
idx, data = self._get_data()
File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 941, in _get_data
success, data = self._try_get_data()
File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 779, in _try_get_data
data = self._data_queue.get(timeout=timeout)
File "/usr/lib/python3.6/multiprocessing/queues.py", line 104, in get
if not self._poll(timeout):
File "/usr/lib/python3.6/multiprocessing/connection.py", line 257, in poll
return self._poll(timeout)
File "/usr/lib/python3.6/multiprocessing/connection.py", line 414, in _poll
r = wait([self], timeout)
File "/usr/lib/python3.6/multiprocessing/connection.py", line 911, in wait
ready = selector.select(timeout)
File "/usr/lib/python3.6/selectors.py", line 376, in select
fd_event_list = self._poll.poll(timeout)
KeyboardInterrupt
~~~

and here is the log of epoch 0 of the second time.

~~~
root@takeshi-jetson:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --data=data/fruit --model-dir=models/fruit --batch-size=4 --epochs=30 |& tee data/fruit/train.log
2021-07-20 13:19:57 - Using CUDA...
2021-07-20 13:19:57 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/fruit', dataset_type='open_images', datasets=['data/fruit'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=30, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2021-07-20 13:19:57 - Prepare training datasets.
2021-07-20 13:19:57 - loading annotations from: data/fruit/sub-train-annotations-bbox.csv
2021-07-20 13:19:57 - annotations loaded from:  data/fruit/sub-train-annotations-bbox.csv
num images:  5145
2021-07-20 13:20:11 - Dataset Summary:Number of Images: 5145
Minimum Number of Images for a Class: -1
Label Distribution:
Apple: 3622
Banana: 1574
Grape: 2560
Orange: 6186
Pear: 757
Pineapple: 534
Strawberry: 7553
Watermelon: 753
2021-07-20 13:20:11 - Stored labels into file models/fruit/labels.txt.
2021-07-20 13:20:11 - Train dataset size: 5145
2021-07-20 13:20:11 - Prepare Validation datasets.
2021-07-20 13:20:11 - loading annotations from: data/fruit/sub-test-annotations-bbox.csv
2021-07-20 13:20:12 - annotations loaded from:  data/fruit/sub-test-annotations-bbox.csv
num images:  930
2021-07-20 13:20:14 - Dataset Summary:Number of Images: 930
Minimum Number of Images for a Class: -1
Label Distribution:
Apple: 329
Banana: 132
Grape: 446
Orange: 826
Pear: 107
Pineapple: 105
Strawberry: 754
Watermelon: 125
2021-07-20 13:20:14 - Validation dataset size: 930
2021-07-20 13:20:14 - Build network.
2021-07-20 13:20:14 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
[... 4 lines cut ...]
/usr/local/lib/python3.6/dist-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
"https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
2021-07-20 13:22:31 - Epoch: 0, Step: 10/1287, Avg Loss: 12.7006, Avg Regression Loss 3.4951, Avg Classification Loss: 9.2055
2021-07-20 13:22:41 - Epoch: 0, Step: 20/1287, Avg Loss: 8.1384, Avg Regression Loss 2.9167, Avg Classification Loss: 5.2217
2021-07-20 13:23:06 - Epoch: 0, Step: 30/1287, Avg Loss: 7.3496, Avg Regression Loss 2.3725, Avg Classification Loss: 4.9770

...

2021-07-20 14:01:47 - Epoch: 0, Step: 1260/1287, Avg Loss: 5.7504, Avg Regression Loss 1.9949, Avg Classification Loss: 3.7555
2021-07-20 14:01:58 - Epoch: 0, Step: 1270/1287, Avg Loss: 5.4125, Avg Regression Loss 1.6238, Avg Classification Loss: 3.7888
2021-07-20 14:02:05 - Epoch: 0, Step: 1280/1287, Avg Loss: 5.8632, Avg Regression Loss 1.9420, Avg Classification Loss: 3.9213
/usr/local/lib/python3.6/dist-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
warnings.warn(warning.format(ret))
2021-07-20 14:03:06 - Epoch: 0, Validation Loss: 5.3136, Validation Regression Loss 1.4593, Validation Classification Loss: 3.8544
2021-07-20 14:03:06 - Saved model models/fruit/mb1-ssd-Epoch-0-Loss-5.31362990657659.pth
2021-07-20 14:03:17 - Epoch: 1, Step: 10/1287, Avg Loss: 5.7705, Avg Regression Loss 1.4058, Avg Classification Loss: 4.3647
[... cut ...]
2021-07-20 14:03:40 - Epoch: 1, Step: 40/1287, Avg Loss: 5.5791, Avg Regression Loss 1.8394, Avg Classification Loss: 3.7397
2021-07-20 14:03:48 - Epoch: 1, Step: 50/1287, Avg Loss: 5.2032, Avg Regression Loss 1.5836, Avg Classification Loss: 3.6195
2021-07-20 14:04:04 - Epoch: 1, Step: 60/1287, Avg Loss: 5.4133, Avg Regression Loss 1.6319, Avg Classification Loss: 3.7814
2021-07-20 14:04:20 - Epoch: 1, Step: 70/1287, Avg Loss: 6.1341, Avg Regression Loss 2.2768, Avg Classification Loss: 3.8573

...
~~~

dusty-nv, 2021-07-20:
Can you try setting `--batch-size 1 --workers 0` when you run `train_ssd.py`?

Also, have SWAP mounted and try disabling the GUI if you haven't already:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

take5553, 2021-07-22:
I executed `train_ssd.py` three times with the parameters you mention and the script seems to work stably although the processing time gets longer. Thanks!
So, do you think the cause of stuck comes from the shortage of memory?

dusty-nv, 2021-07-22:
> So, do you think the cause of stuck comes from the shortage of memory?

If those arguments helped, yes I think the cause may have been memory-related.  Since it was getting stuck in the dataloader, you can try keeping `--workers 0` but increasing `--batch-size 2`.   The `--workers` sets the number of dataloaders (0 disables parallel dataloading)

closed 2022-04-15 by Fkjmp