dusty-nv/jetson-inference #784: Structure for Creating Subclasses for Image Classification

GOBish, 2020-10-30:
Hi @dusty-nv ,
Thanks again for the great tutorials.
I am trying to create a custom image classification model that has subclasses. Could you demonstrate what the file structure and labels.txt format should look like?

Thanks!

dusty-nv, 2020-10-31:
Hi GOBish, this document shows the structure of classification dataset:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md

GOBish, 2020-10-31:
Thanks @dusty-nv — I have used that successfully for creating object classification and it works well.
Do you have any examples, though, related to my question above of having sub-classes?
Thanks!

dusty-nv, 2020-10-31:
I haven't seen that supported with ResNet, GoogleNet, ect.   They just support N number of classes.  Additional structure you would have to store yourself in another file (i.e. Golden Retriever is Dog)

GOBish, 2020-10-31:
Ok, got it.
Do you think there is a way to approach it with multi-label classifying, as described here:
https://www.learnopencv.com/multi-label-image-classification-with-pytorch/

dusty-nv, 2020-11-01:
That article is using a model that they modified to do multi-label classification- you would need to export it from PyTorch to ONNX and see if you could load it with trtexec (from /usr/src/tensorrt/bin).

Then you would have to modify imageNet C++ class to interpret the multiple outputs.

Of course, you could just run it in PyTorch on the Jetson if the PyTorch performance is good enough for you.

https://www.learnopencv.com/multi-label-image-classification-with-pytorch/

GOBish, 2020-11-01:
Got it! Thanks as always @dusty-nv

closed 2020-11-01 by GOBish