dusty-nv/jetson-inference #786: Syntax for detectnet, when refering to a custom trained model

ebachs10, 2020-11-04:
Hi,

First of all thanks for the very nice library.

I'm trying to execute detectnet using my own custom "coin" model in python. I have the example working with:

detectnet --model=models/coin/ssd-mobilenet.onnx --labels=models/coin/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            csi://0

where the current folder is _jetson-inference/python/training/detection/ssd_

Also I have a working python script from the tutorial:

import jetson.inference
import jetson.utils

# load the recognition network
jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)

My question:
I can not figure out, how to set my custom "coin" model  as netwok in the python scrip, like
jetson.inference.detectNet("coins", threshold=0.5)

Can somebody help me out with the correct syntax? The python scirpt is executed in another folder than the jetson-inference.

Thanks in advance

/Esben

dusty-nv, 2020-11-04:
Hi @ebachs10 , you should be able to do it like this:

``` python
net = jetson.inference.detectNet(argv=[‘--model=models/coin/ssd-mobilenet.onnx’, ‘--labels=models/coin/labels.txt’, ‘--input-blob=input_0’, ‘--output-cvg=scores’, ‘--output-bbox=boxes’], threshold=0.5)
```

w1ngedshadow, 2020-11-05:
@dusty-nv @ebachs10  Hello there, I modified this string to this:
`net = jetson.inference.detectNet(argv=["--model=/home/user/jetson-inference/python/training/detection/ssd/models/myModel/ssd-mobilenet.onnx", "--labels=/home/user/jetson-inference/python/training/detection/ssd/models/myModel/labels.txt", "--input-blob=input_0", "--output-cvg=scores", "--output-bbox=boxes"], threshold=0.5)`
It doesnt work for me with ‘ ’  and without full path to model/labels.

ebachs10, 2020-11-05:
@w1ngedshadow  working like a champ! To @dusty-nv and @w1ngedshadow  thank you for the quick answers!

raghuram9581, 2021-02-23:
import jetson.inference
import jetson.utils
import argparse
import sys

net = jetson.inference.detectNet(argv=["model=models/flowers/ssd-mobilenet.onnx", "labels=models/flowers/labels.txt", "input-blob=input_0", "output-cvg=scores", "output-bbox=boxes"],threshold=0.5)

camera = jetson.utils.videoSource("/dev/video0")      # '/dev/video0' for V4L2
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))

when i am using the about given commad for loading the custom .onnx model, it is loading default mobilenetv2 model not the custom model i have given.

please help me.

console output:

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

dusty-nv, 2021-02-23:
Hi @raghuram9581, put `--` in front of the arguments, like this:

``` python
net = jetson.inference.detectNet(argv=["--model=models/flowers/ssd-mobilenet.onnx", "--labels=models/flowers/labels.txt", "--input-blob=input_0", "--output-cvg=scores", "--output-bbox=boxes"],threshold=0.5)
```

karimmig, 2022-11-16:
hello @dusty-nv  could you help me, so i'm working on distract driver detection and i have trained a model and convert it into ONNX, and now i want to know how to deploy it into my jetson nano for live streaming

dusty-nv, 2022-11-16:
Hi @karimmig, did you train the model with train_ssd.py?   detectnet is setup to use those kind of custom models

closed 2023-04-27 by dusty-nv