dusty-nv/jetson-inference #787: Issue with building repo from source on "-j$(nproc)" command.

DJT777, 2020-11-06:
Hello!

I am building the repository from source on a Jetson TX1 using Jetpack 4.4. So far I have succesfully run these commands.

`sudo apt-get update
$ sudo apt-get install git cmake libpython3-dev python3-numpy
$ git clone --recursive https://github.com/dusty-nv/jetson-inference
$ cd jetson-inference
$ mkdir build
$ cd build
$ cmake ../`

but when I arrive to this command

`make -j$(nproc)`

I am receiving this in the terminal
`[  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFilterMode.cu.o
[  3%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
[  3%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o
[  3%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaCrop.cu.o
/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/dylan/disk/HelloAIWorld/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined
[... 58 lines cut ...]
  /media/dylan/disk/HelloAIWorld/jetson-inference/build/utils/CMakeFiles/jetson-utils.dir/cuda/./jetson-utils_generated_cudaCrop.cu.o

utils/CMakeFiles/jetson-utils.dir/build.make:524: recipe for target 'utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaCrop.cu.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaCrop.cu.o] Error 1
12 errors detected in the compilation of "/tmp/tmpxft_000003e6_00000000-8_cudaYUV-YV12.compute_72.cpp1.ii".
CMake Error at jetson-utils_generated_cudaYUV-YV12.cu.o.cmake:279 (message):
  Error generating file
  /media/dylan/disk/HelloAIWorld/jetson-inference/build/utils/CMakeFiles/jetson-utils.dir/cuda/./jetson-utils_generated_cudaYUV-YV12.cu.o

utils/CMakeFiles/jetson-utils.dir/build.make:1072: recipe for target 'utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o] Error 1
12 errors detected in the compilation of "/tmp/tmpxft_000003f5_00000000-8_cudaColormap.compute_72.cpp1.ii".
CMake Error at jetson-utils_generated_cudaColormap.cu.o.cmake:279 (message):
  Error generating file
  /media/dylan/disk/HelloAIWorld/jetson-inference/build/utils/CMakeFiles/jetson-utils.dir/cuda/./jetson-utils_generated_cudaColormap.cu.o

utils/CMakeFiles/jetson-utils.dir/build.make:290: recipe for target 'utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o] Error 1
CMakeFiles/Makefile2:713: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
`

I am wondering if you can help me find a solution to this

DJT777, 2020-11-06:
UPDATE:
I have attempted to use docker, but found another problem. If you look below you will see that my jetson's mmcblkop1 partition is indeed full. I am not attempting to work in that partition. 

The partition these files are being worked in are in mmcblk2p1, where I have over 25 gigabytes of free space, yet I get this:

```
dylan@tx1:/media/dylan/disk/HelloAIWorld/jetson-inference$ bash docker/run.sh --volume /media/dylan/disk/HelloAIWorld/jetson-inference/:media/dylan/disk/HelloAIWorld
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.4.4
size of data/networks:  108069 bytes
Hit:1 https://repo.download.nvidia.com/jetson/common r32.4 InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease  
Hit:4 https://repo.download.nvidia.com/jetson/t210 r32.4 InRelease   
Hit:5 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Reading package lists... Done                     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
dialog is already the newest version (1.3-20171209-1).
The following package was automatically installed and is no longer required:
  libgtkmm-2.4-1v5
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 55 not upgraded.
Models have not yet been downloaded, running model downloader tool now...
docker/run.sh: line 68: ./download-models.sh: Permission denied
Downloading pytorch-ssd base model...
python/training/d 100%[===========>]  36.23M  3.74MB/s    in 8.4s    
CONTAINER:     dustynv/jetson-inference:r32.4.4
DATA_VOLUME:   --volume /media/dylan/disk/HelloAIWorld/jetson-inference/data:/jetson-inference/data --volume /media/dylan/disk/HelloAIWorld/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /media/dylan/disk/HelloAIWorld/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /media/dylan/disk/HelloAIWorld/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /media/dylan/disk/HelloAIWorld/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:    -v /media/dylan/disk/HelloAIWorld/jetson-inference/:media/dylan/disk/HelloAIWorld 
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
Unable to find image 'dustynv/jetson-inference:r32.4.4' locally
r32.4.4: Pulling from dustynv/jetson-inference
296c9ad75bee: Pulling fs layer 
c0533d139302: Pulling fs layer 
3c11bb34abc8: Pulling fs layer 
45e266ea8946: Pulling fs layer 
3f9fe4230f83: Pulling fs layer 
45e266ea8946: Extracting  88.57MB/164.8MB
a7c06ca5da1d: Download complete 
21a2a17ba5e5: Download complete 
fa6220c842e3: Download complete 
5b7f5976ffd4: Download complete 
18e107db1fc8: Download complete 
3e74e1a748d9: Download complete 
38d909b2f33b: Download complete 
2b80aba2718f: Download complete 
b2ddf8a7ddf1: Download complete 
ed1d76c84ec3: Download complete 
08e440b62d01: Download complete 
3e5d9460f96f: Download complete 
f313550a9e60: Downloading  118.3MB/118.3MB
a742b8e794d7: Download complete 
628353978f08: Download complete 
7adc69170422: Downloading  275.2MB/275.2MB
e0d1fc183e52: Download complete 
948ac0abde32: Downloading  28.61MB/28.61MB
adfb60e46ee1: Waiting 
863e90653906: Waiting 
3e9098193eb0: Waiting 
c8725fd1629c: Waiting 
97d7d811f1b0: Waiting 
187638a96698: Waiting 
7131073b404c: Waiting 
2468aca51358: Waiting 
597dd74d4b4e: Waiting 
27390dd65acc: Waiting 
9c42f0b824ae: Waiting 
4500dbbd2046: Waiting 
7a2b3e62c40b: Waiting 
40131d437449: Waiting 
a5c652547df0: Waiting 
9f2cc1d0a735: Waiting 
4c296bd90c8d: Waiting 
ce3e627ba715: Waiting 
4544c0f76afb: Waiting 
docker: write /var/lib/docker/tmp/GetImageBlob388966471: no space left on device.
See 'docker run --help'.
dylan@tx1:/media/dylan/disk/HelloAIWorld/jetson-inference$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1   14G   13G   61M 100% /
none            1.7G     0  1.7G   0% /dev
tmpfs           2.0G  247M  1.7G  13% /dev/shm
tmpfs           2.0G   96M  1.9G   5% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs           396M  132K  396M   1% /run/user/1000
/dev/mmcblk2p1   30G  1.5G   28G   6% /media/dylan/disk
```
I am wondering if you can help me solve this.

dusty-nv, 2020-11-06:
Hi @DJT777 , by default Docker will still try to download/store container images to your system partition.

To change where Docker stores images, set the `"data-root"` variable in your `/etc/docker/daemon.json` config file:

``` json
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

    "default-runtime": "nvidia",
    "data-root": "/media/dylan/disk/docker"
}
```

Also run `mkdir /media/dylan/disk/docker` and make sure that your secondary partition will be mounted at boot (i.e. by adding it to `/etc/fstab`)

Then reboot your system, and check `sudo docker info` to confirm that the Docker data dir is on the new path you created.  The container image will then be downloaded to that new path.

DJT777, 2020-11-07:
So I did my best to follow your instructions, and I am getting an error that L4T is not up to date when I am on 4.4. I've provided the terminal output while I've been working on this as well as the text for my daemon.json and fstab. 

I am assuming that build.sh is contained within the partition and is not seeing the other partition and thus does not recognize the version of L4T. I would like your help on this because I would prefer to follow your instructions than crash my Jetson while I am problem solving. Please have a look.

**daemon.js**
```
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

    "default-runtime": "nvidia",
    "data-root": "/media/dylan/disk/docker"
}
```

**fstab**```

# /etc/fstab: static file system information.
#
# These are the filesystems that are always mounted on boot, you can
# override any of these by copying the appropriate line from this file into
# /etc/fstab and tweaking it as you see fit.  See fstab(5).
#
# <file system> <mount point>             <type>          <options>                               <dump> <pass>
/dev/root            /                     ext4           defaults                                     0 1
/dev/mmcblk2p1 /media/dylan/disk/docker vfat
```

**build.sh and checking L4T Version**
```
dylan@tx1:/media/dylan/disk/docker/HelloAIWorld/jetson-inference/docker$ bash build.sh
build.sh: line 16: tools/l4t-version.sh: No such file or directory
build.sh: line 19: [: =: unary operator expected
build.sh: line 21: [: =: unary operator expected
build.sh: line 23: [: =: unary operator expected
cannot build jetson-inference docker container for L4T R
please upgrade to the latest JetPack, or build jetson-inference natively
dylan@tx1:/media/dylan/disk/docker/HelloAIWorld/jetson-inference/docker$ head -n 1 /etc/nv_tegra_release
# R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t210ref, EABI: aarch64, DATE: Fri Oct 16 19:44:43 UTC 2020
dylan@tx1:/media/dylan/disk/docker/HelloAIWorld/jetson-inference/docker$ sudo docker info
[sudo] password for dylan: 
Client:
 Debug Mode: false

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 0
 Server Version: 19.03.6
 Storage Driver: overlay2
  Backing Filesystem: <unknown>
  Supports d_type: true
  Native Overlay Diff: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: nvidia runc
 Default Runtime: nvidia
 Init Binary: docker-init
 containerd version: 
 runc version: 
 init version: 
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 4.9.140-tegra
 Operating System: Ubuntu 18.04.5 LTS
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 3.863GiB
 Name: tx1
 ID: YGTM:HWPE:DAN7:D2A2:GTYV:WN2M:SE4X:3P66:6OJO:WQ2K:TBSH:Q3RP
 Docker Root Dir: /media/dylan/disk/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

```

**df -Th**
```

Filesystem     Type      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1 ext4       14G   13G   81M 100% /
none           devtmpfs  1.7G     0  1.7G   0% /dev
tmpfs          tmpfs     2.0G   67M  1.9G   4% /dev/shm
tmpfs          tmpfs     2.0G   21M  2.0G   2% /run
tmpfs          tmpfs     5.0M  4.0K  5.0M   1% /run/lock
tmpfs          tmpfs     2.0G     0  2.0G   0% /sys/fs/cgroup
/dev/mmcblk2p1 vfat       30G  1.5G   28G   6% /media/dylan/disk/docker
tmpfs          tmpfs     396M  124K  396M   1% /run/user/1000

```

DJT777, 2020-11-07:
UPDATE:

I am now able to get build.sh to run properly. Space is no longer and issue because I have changed the settings you instructed. However I am now getting this:

```
sudo bash docker/build.sh
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.4.4
BASE_IMAGE=nvcr.io/nvidia/l4t-pytorch:r32.4.4-pth1.6-py3
TAG=jetson-inference:r32.4.4
Sending build context to Docker daemon  207.3MB
Step 1/18 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-pytorch:r32.4.3-pth1.6-py3
Step 2/18 : FROM ${BASE_IMAGE}
r32.4.4-pth1.6-py3: Pulling from nvidia/l4t-pytorch
e74fe6ef6bd6: Pulling fs layer 
7dcdd1c8f1d2: Pulling fs layer 
148ea20d31e0: Pulling fs layer 
fbc4cd4d050b: Pulling fs layer 
a21b0b3d8206: Pulling fs layer 
4ba0c94f9855: Pulling fs layer 
fbc4cd4d050b: Download complete 
84ff17ad4b18: Download complete 
5ac903fdc4a8: Download complete 
ecf00917e120: Download complete 
30d000a9cd22: Download complete 
a26b515ffe8f: Download complete 
a199cb2dd71e: Download complete 
c4f4e0f882d3: Download complete 
3e956de9ea4b: Download complete 
e26b78d1aaed: Download complete 
ac42496d0bc2: Download complete 
7db2983f5802: Download complete 
f313550a9e60: Download complete 
a742b8e794d7: Download complete 
628353978f08: Download complete 
7adc69170422: Download complete 
e0d1fc183e52: Download complete 
948ac0abde32: Download complete 
adfb60e46ee1: Download complete 
863e90653906: Download complete 
3e9098193eb0: Download complete 
c8725fd1629c: Download complete 
97d7d811f1b0: Download complete 
failed to register layer: symlink ../bfb8cec0cb03b81f38f63cd73707504ef3894bac7bde9d3372e3c0cec0f773cd/diff /media/dylan/disk/docker/overlay2/l/YLRLHWMXVQL5JTLC5CUWMMBXJ7: operation not permitted
```

dusty-nv, 2020-11-07:
Run the script from jetson-inference directory, like this:

$ cd /path/to/your/jetson-inference
$ docker/run.sh

DJT777, 2020-11-07:
I got the same error. This time by running run.sh I got errors in downloading the models.

```
/media/dylan/disk/docker/HelloAIWorld/jetson-inference$ bash /docker/run.sh

Hello AI World (jetson-inference)
 ──────────────────────────────────────────────────────────────────────────────

          ┌─────────────────────Download Error───────────────────────┐
          │                                                          │
          │ Failed to download                                       │
          │ 'FCN-ResNet18-SUN-RGBD-512x400.tar.gz' (error code=1)    │
          │                                                          │
          │ Would you like to try downloading it again?              │
          │                                                          │
          │ Note:  if this error keeps occuring, see here:           │
          │  https://github.com/dusty-nv/jetson-inference/releases   │
          ├──────────────────────────────────────────────────────────┤
          │           <Retry>      <Next >      <Quit >              │
          └──────────────────────────────────────────────────────────┘

 
[jetson-inference]  to run this tool again, use the following commands:
 
    $ cd <jetson-inference>/tools
    $ ./download-models.sh
 
Downloading pytorch-ssd base model...
python/training/detection/ssd/models/mobilenet-v1-ssd-mp-0_675.pth: Permission denied
CONTAINER:     dustynv/jetson-inference:r32.4.4
DATA_VOLUME:   --volume /media/dylan/disk/docker/HelloAIWorld/jetson-inference/data:/jetson-inference/data --volume /media/dylan/disk/docker/HelloAIWorld/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /media/dylan/disk/docker/HelloAIWorld/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /media/dylan/disk/docker/HelloAIWorld/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /media/dylan/disk/docker/HelloAIWorld/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
Unable to find image 'dustynv/jetson-inference:r32.4.4' locally
^[[Cr32.4.4: Pulling from dustynv/jetson-inference
296c9ad75bee: Pulling fs layer 
c0533d139302: Pulling fs layer 
3c11bb34abc8: Pulling fs layer 
45e266ea8946: Pulling fs layer 
3f9fe4230f83: Pulling fs layer 
2331551ae91f: Pulling fs layer 
a7c06ca5da1d: Pulling fs layer 
21a2a17ba5e5: Pulling fs layer 
fa6220c842e3: Pulling fs layer 
5b7f5976ffd4: Pulling fs layer 
18e107db1fc8: Pulling fs layer 
3e74e1a748d9: Pulling fs layer 
38d909b2f33b: Pulling fs layer 
2b80aba2718f: Pulling fs layer 
b2ddf8a7ddf1: Pulling fs layer 
ed1d76c84ec3: Pulling fs layer 
08e440b62d01: Pulling fs layer 
3e5d9460f96f: Pulling fs layer 
f313550a9e60: Pulling fs layer 
a742b8e794d7: Pulling fs layer 
c0533d139302: Download complete 
7adc69170422: Download complete 
e0d1fc183e52: Download complete 
948ac0abde32: Download complete 
adfb60e46ee1: Download complete 
863e90653906: Download complete 
3e9098193eb0: Download complete 
c8725fd1629c: Download complete 
97d7d811f1b0: Download complete 
187638a96698: Download complete 
7131073b404c: Download complete 
2468aca51358: Download complete 
597dd74d4b4e: Download complete 
27390dd65acc: Download complete 
9c42f0b824ae: Download complete 
4500dbbd2046: Download complete 
7a2b3e62c40b: Download complete 
40131d437449: Download complete 
a5c652547df0: Download complete 
9f2cc1d0a735: Download complete 
4c296bd90c8d: Download complete 
ce3e627ba715: Download complete 
4544c0f76afb: Download complete 
docker: failed to register layer: symlink ../f2864d647818c72b31c7cc4210ab08bb88b7f6fa77aca1fa68ed57c721e65a9d/diff /media/dylan/disk/docker/overlay2/l/C5IBBIDKGN6SI7CJF62IYWM6MK: operation not permitted.
See 'docker run --help'.

```

dusty-nv, 2020-11-07:
Don't put your jetson-inference directory in that docker data directory, because docker owns the permissions.

Reclone the repo into another directory on your secondary permission outside of that docker dir.

          │  https://github.com/dusty-nv/jetson-inference/releases   │

          ├──────────────────────────────────────────────────────────┤

DJT777, 2020-11-07:
The docket folder contents are in the SD card. I figure I should just add another folder into the path in fstab and daemon.json.

I.e.

 "/media/dylan/disk/NEW_FOLDER/docker"

Instead of 
 "/media/dylan/disk/docker"

DJT777, 2020-11-08:
I reflashed my Jetson, reformatted the SD card, and reorganized the docker mount and adjusted the file system.

build.sh
```
docker: failed to register layer: symlink ../0f07b6a9cbd1594244ca531ed2f8cfc989ea0665fb67e7b580cc77301cbaa649/diff /media/user1/D6E1-674D/disk/docker/overlay2/l/PAYAGOOVCWJXNMMAPVDIDHSRVL: operation not permitted.
See 'docker run --help'.
user1@jetsontx1:/media/user1/D6E1-674D/disk/jetson-inference$ cd ..
user1@jetsontx1:/media/user1/D6E1-674D/disk$ ls
docker  jetson-inference
user1@jetsontx1:/media/user1/D6E1-674D/disk$ 

```

DJT777, 2020-11-08:
I have also attempted to build from source again, but get this error. This is very strange.

```
user1@jetsontx1:/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build$ sudo make -j$(nproc)
[  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
[  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaCrop.cu.o
[  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFilterMode.cu.o
[  3%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o
/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

[... 43 lines cut ...]
CMake Error at jetson-utils_generated_cudaCrop.cu.o.cmake:279 (message):
  Error generating file
[... cut ...]
  /media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/utils/CMakeFiles/jetson-utils.dir/cuda/./jetson-utils_generated_cudaColormap.cu.o

utils/CMakeFiles/jetson-utils.dir/build.make:63: recipe for target 'utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o] Error 1
CMakeFiles/Makefile2:713: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

dusty-nv, 2020-11-09:
> /media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

It is strange to get that error - can you try adding `#include <stdint.h>` to the top of your `jetson-inference/utils/network/Endian.h` file?

Then re-run:

``` bash
cd /media/user1/D6E1-674D/disk/fromsource/jetson-inference/build
cmake ../
make
sudo make install
```

dusty-nv, 2020-11-09:
> docker: failed to register layer: symlink ../0f07b6a9cbd1594244ca531ed2f8cfc989ea0665fb67e7b580cc77301cbaa649/diff /media/user1/D6E1-674D/disk/docker/overlay2/l/PAYAGOOVCWJXNMMAPVDIDHSRVL: operation not permitted.

Is your disk being mounted at boot time in `/etc/fstab`?

Can you try running `sudo docker system prune --volumes -a` and then restarting your system?     Then try this:

``` bash
$ cd /media/user1/D6E1-674D/disk/fromsource/jetson-inference
$ docker/run.sh
```

If it still doesn't work, please post the output of the following commands:

* `ls -ll /media/user1/D6E1-674D/disk/`
* `sudo docker info`
* `cat /etc/docker/daemon.json`

DJT777, 2020-11-09:
Output
```
user1@jetsontx1:/media/user1/D6E1-674D/disk/fromsource/jetson-inference$ ls -ll /media/user1/D6E1-674D/disk/
total 48
drwxr-xr-x 14 root root 16384 Nov  9 15:51 docker
drwxr-xr-x  3 root root 16384 Nov  7 20:26 fromsource
drwxr-xr-x 13 root root 16384 Nov  7 20:02 jetson-inference
user1@jetsontx1:/media/user1/D6E1-674D/disk/fromsource/jetson-inference$ sudo docker info
Client:
 Debug Mode: false

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 0
 Server Version: 19.03.6
 Storage Driver: overlay2
  Backing Filesystem: <unknown>
  Supports d_type: true
  Native Overlay Diff: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: nvidia runc
 Default Runtime: nvidia
 Init Binary: docker-init
 containerd version: 
 runc version: 
 init version: 
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 4.9.140-tegra
 Operating System: Ubuntu 18.04.5 LTS
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 3.863GiB
 Name: jetsontx1
 ID: GP7Q:JUYX:D6BE:ESCP:F52F:MXEJ:PNZ7:XFJ3:XCO4:CCSL:DE4I:5Z6I
 Docker Root Dir: /media/user1/D6E1-674D/disk/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

user1@jetsontx1:/media/user1/D6E1-674D/disk/fromsource/jetson-inference$ cat /etc/docker/daemon.json
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

    "default-runtime": "nvidia",
    "data-root": "/media/user1/D6E1-674D/disk/docker"
}
user1@jetsontx1:/media/user1/D6E1-674D/disk/fromsource/jetson-inference$ 

```

DJT777, 2020-11-09:
Also,

the same error after editing Endian.sh
```
user1@jetsontx1:/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build$ sudo make
[  1%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(40): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(41): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(42): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(50): error: identifier "uint64_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(51): error: identifier "uint32_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

/media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/aarch64/include/jetson-utils/endian.h(52): error: identifier "uint16_t" is undefined

12 errors detected in the compilation of "/tmp/tmpxft_00002976_00000000-8_cudaYUV-YV12.compute_72.cpp1.ii".
CMake Error at jetson-utils_generated_cudaYUV-YV12.cu.o.cmake:279 (message):
  Error generating file
  /media/user1/D6E1-674D/disk/fromsource/jetson-inference/build/utils/CMakeFiles/jetson-utils.dir/cuda/./jetson-utils_generated_cudaYUV-YV12.cu.o

utils/CMakeFiles/jetson-utils.dir/build.make:1072: recipe for target 'utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o' failed
make[2]: *** [utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o] Error 1
CMakeFiles/Makefile2:713: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

dusty-nv, 2020-11-09:
Ahh...i notice in your fstab, your 2nd partition is vfat.  This could explain why docker fails to make symbolic links?

Can you reformat your 2nd disk as ext4?

DJT777, 2020-11-13:
Took me a couple days to get back to this.

I have reflashed, reformatted the SD as ext4, and it is mounted in fstab. I had to generate a new daemon.json.

Now I am getting this:

```
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.4.4
size of data/networks:  535209937 bytes
CONTAINER:     dustynv/jetson-inference:r32.4.4
DATA_VOLUME:   --volume /media/dylan/SD/jetson-inference/data:/jetson-inference/data --volume /media/dylan/SD/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /media/dylan/SD/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /media/dylan/SD/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /media/dylan/SD/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
docker: Error response from daemon: Unknown runtime specified nvidia.
See 'docker run --help'.

```

DJT777, 2020-11-13:
Some more weird errors after creating and setting daemon.json to default settings:

```
dylan@tx1:/media/dylan/SD/jetson-inference$ sudo docker/run.sh
[sudo] password for dylan: 
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.4.4
size of data/networks:  535209937 bytes
CONTAINER:     dustynv/jetson-inference:r32.4.4
DATA_VOLUME:   --volume /media/dylan/SD/jetson-inference/data:/jetson-inference/data --volume /media/dylan/SD/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /media/dylan/SD/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /media/dylan/SD/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /media/dylan/SD/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
Unable to find image 'dustynv/jetson-inference:r32.4.4' locally
r32.4.4: Pulling from dustynv/jetson-inference
296c9ad75bee: Pulling fs layer 
c0533d139302: Pulling fs layer 
3c11bb34abc8: Pulling fs layer 
45e266ea8946: Pulling fs layer 
3f9fe4230f83: Pulling fs layer 
2331551ae91f: Pulling fs layer 
a7c06ca5da1d: Pulling fs layer 
21a2a17ba5e5: Pulling fs layer 
fa6220c842e3: Pulling fs layer 
5b7f5976ffd4: Pulling fs layer 
18e107db1fc8: Pulling fs layer 
3e74e1a748d9: Pulling fs layer 
38d909b2f33b: Pulling fs layer 
2b80aba2718f: Pulling fs layer 
b2ddf8a7ddf1: Pulling fs layer 
ed1d76c84ec3: Pulling fs layer 
08e440b62d01: Pulling fs layer 
3e5d9460f96f: Pulling fs layer 
3e5d9460f96f: Waiting 
a742b8e794d7: Pulling fs layer 
628353978f08: Pull complete 
7adc69170422: Pull complete 
e0d1fc183e52: Pull complete 
948ac0abde32: Pull complete 
adfb60e46ee1: Pull complete 
863e90653906: Pull complete 
3e9098193eb0: Pull complete 
c8725fd1629c: Pull complete 
97d7d811f1b0: Pull complete 
187638a96698: Pull complete 
7131073b404c: Pull complete 
2468aca51358: Pull complete 
597dd74d4b4e: Pull complete 
27390dd65acc: Pull complete 
9c42f0b824ae: Pull complete 
4500dbbd2046: Pull complete 
7a2b3e62c40b: Pull complete 
40131d437449: Pull complete 
a5c652547df0: Pull complete 
9f2cc1d0a735: Pull complete 
4c296bd90c8d: Pull complete 
ce3e627ba715: Pull complete 
4544c0f76afb: Pull complete 
Digest: sha256:d6261070228943bf0f5e5f6cd9123dfb0800ecf3953941521095edef8e4ce721
Status: Downloaded newer image for dustynv/jetson-inference:r32.4.4
docker: Error response from daemon: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v1.linux/moby/6270fea6592768722f254604159169329f79066f493dd63ad3e08091b403ee0e/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: : unknown.

```

DJT777, 2020-11-13:
Took a couple days to get back to working on this.

I've got the SD card formatted and mounted in fstab as an ext4 partition. The daemon.js is also as you recommended.

```
dylan@tx1:/media/dylan/SD/jetson-inference$ sudo docker/run.sh 
[sudo] password for dylan: 
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.4.4
size of data/networks:  535209937 bytes
CONTAINER:     dustynv/jetson-inference:r32.4.4
DATA_VOLUME:   --volume /media/dylan/SD/jetson-inference/data:/jetson-inference/data --volume /media/dylan/SD/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /media/dylan/SD/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /media/dylan/SD/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /media/dylan/SD/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
Unable to find image 'dustynv/jetson-inference:r32.4.4' locally
r32.4.4: Pulling from dustynv/jetson-inference
296c9ad75bee: Pulling fs layer 
c0533d139302: Pulling fs layer 
3c11bb34abc8: Pulling fs layer 
45e266ea8946: Pulling fs layer 
3f9fe4230f83: Pulling fs layer 
2331551ae91f: Pulling fs layer 
a7c06ca5da1d: Pulling fs layer 
21a2a17ba5e5: Pulling fs layer 
fa6220c842e3: Pulling fs layer 
5b7f5976ffd4: Pulling fs layer 
18e107db1fc8: Pulling fs layer 
3e74e1a748d9: Pulling fs layer 
38d909b2f33b: Pulling fs layer 
2b80aba2718f: Pulling fs layer 
b2ddf8a7ddf1: Pulling fs layer 
ed1d76c84ec3: Pulling fs layer 
08e440b62d01: Pulling fs layer 
3e5d9460f96f: Pulling fs layer 
f313550a9e60: Pulling fs layer 
a742b8e794d7: Pulling fs layer 
628353978f08: Pull complete 
7adc69170422: Pull complete 
e0d1fc183e52: Pull complete 
948ac0abde32: Pull complete 
adfb60e46ee1: Pull complete 
863e90653906: Pull complete 
3e9098193eb0: Pull complete 
c8725fd1629c: Pull complete 
97d7d811f1b0: Pull complete 
187638a96698: Pull complete 
7131073b404c: Pull complete 
2468aca51358: Pull complete 
597dd74d4b4e: Pull complete 
27390dd65acc: Pull complete 
9c42f0b824ae: Pull complete 
4500dbbd2046: Pull complete 
7a2b3e62c40b: Pull complete 
40131d437449: Pull complete 
a5c652547df0: Pull complete 
9f2cc1d0a735: Pull complete 
4c296bd90c8d: Pull complete 
ce3e627ba715: Pull complete 
4544c0f76afb: Pull complete 
Digest: sha256:d6261070228943bf0f5e5f6cd9123dfb0800ecf3953941521095edef8e4ce721
Status: Downloaded newer image for dustynv/jetson-inference:r32.4.4
docker: Error response from daemon: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v1.linux/moby/dfa87d7527b7bb2d64c7a5e3a1b084c4340ca92782d5f25cd83bd93fae71279d/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: : unknown.
dylan@tx1:/media/dylan/S

```

dusty-nv, 2020-11-13:
What does your `/etc/docker/daemon.json` file look like now?   Here is what mine looks like:

``` json
cat /etc/docker/daemon.json
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

    "default-runtime": "nvidia",
    "data-root": "/media/nvidia/WD_NVME/docker/data/JetPack_4.4.1/b49/data"
}
```

From the errors you are getting about the nvidia-container-runtime, it appears you might be missing that part from your daemon.json now.

DJT777, 2020-11-16:
daemon.json

```
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    }
}

```

DJT777, 2020-11-16:
I ran it a second time and I am getting a second error.

`docker: failed to register layer: Error processing tar file(exit status 1): write /usr/local/lib/python3.6/dist-packages/torch/lib/libtorch_cuda.so: no space left on device.`

DJT777, 2020-11-16:
I was able to build from source succesfully. I will keep this open for now in casearch I have problems.

dusty-nv, 2020-11-16:
OK gotcha - glad you were eventually able to get it working 👍 

> docker: failed to register layer: Error processing tar file(exit status 1): write /usr/local/lib/python3.6/dist-packages/torch/lib/libtorch_cuda.so: no space left on device.

Regarding your docker error, I think you need to add the `"data-root": "/media/nvidia/WD_NVME/docker/data/JetPack_4.4.1/b49/data"` back into your daemon.json

mohanRajPartha, 2021-07-15:
Hi,
I am building the repository in Jetson Xavier NX using Jetpack 4.5.1,
when i run this command i get the same error. pls help.

```
  0%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFilterMode.cu.o
[  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaColormap.cu.o
[  2%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YV12.cu.o
[  3%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaCrop.cu.o
[  4%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaFont.cu.o
[  5%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaGrayscale.cu.o
[  6%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaNormalize.cu.o
[  7%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaOverlay.cu.o
[  8%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaPointCloud.cu.o
[  8%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaRGB.cu.o
[  9%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaResize.cu.o
[ 10%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-affine.cu.o
[ 11%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-fisheye.cu.o
[ 12%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaWarp-intrinsic.cu.o
[ 13%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-NV12.cu.o
[ 13%] Building NVCC (Device) object utils/CMakeFiles/jetson-utils.dir/cuda/jetson-utils_generated_cudaYUV-YUYV.cu.o
Scanning dependencies of target jetson-utils
[ 15%] Building CXX object utils/CMakeFiles/jetson-utils.dir/URI.cpp.o
[ 15%] Building CXX object utils/CMakeFiles/jetson-utils.dir/XML.cpp.o
[ 13%] Building CXX object utils/CMakeFiles/jetson-utils.dir/filesystem.cpp.o
[ 16%] Building CXX object utils/CMakeFiles/jetson-utils.dir/commandLine.cpp.o
[ 17%] Building CXX object utils/CMakeFiles/jetson-utils.dir/logging.cpp.o
[ 18%] Building CXX object utils/CMakeFiles/jetson-utils.dir/timespec.cpp.o
[ 19%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/gstCamera.cpp.o
[ 20%] Building CXX object utils/CMakeFiles/jetson-utils.dir/camera/v4l2Camera.cpp.o
[ 21%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstDecoder.cpp.o
[ 21%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstEncoder.cpp.o
[ 22%] Building CXX object utils/CMakeFiles/jetson-utils.dir/codec/gstUtility.cpp.o
[ 23%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaBayer.cpp.o
[ 24%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaColorspace.cpp.o
[ 25%] Building CXX object utils/CMakeFiles/jetson-utils.dir/cuda/cudaPointCloud.cpp.o
[ 25%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glBuffer.cpp.o
[ 26%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glCamera.cpp.o
[ 27%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glDisplay.cpp.o
[ 28%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glEvents.cpp.o
[ 29%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glTexture.cpp.o
[ 30%] Building CXX object utils/CMakeFiles/jetson-utils.dir/display/glWidget.cpp.o
[ 30%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageIO.cpp.o
[ 31%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageLoader.cpp.o
[ 32%] Building CXX object utils/CMakeFiles/jetson-utils.dir/image/imageWriter.cpp.o
[ 33%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devInput.cpp.o
[ 34%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devJoystick.cpp.o
[ 34%] Building CXX object utils/CMakeFiles/jetson-utils.dir/input/devKeyboard.cpp.o
[ 35%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/IPv4.cpp.o
[ 36%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/NetworkAdapter.cpp.o
[ 37%] Building CXX object utils/CMakeFiles/jetson-utils.dir/network/Socket.cpp.o
[ 38%] Building CXX object utils/CMakeFiles/jetson-utils.dir/threads/Process.cpp.o
[ 39%] Building CXX object utils/CMakeFiles/jetson-utils.dir/threads/Thread.cpp.o
[ 39%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoOptions.cpp.o
[ 40%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoOutput.cpp.o
[ 41%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoSource.cpp.o
[ 42%] Linking CXX shared library ../aarch64/lib/libjetson-utils.so
[ 42%] Built target jetson-utils
[ 42%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_tensorConvert.cu.o
[ 42%] Building NVCC (Device) object utils/display/gl-display-test/CMakeFiles/gl-display-test.dir/gl-display-test_generated_cudaInteropKernels.cu.o
Scanning dependencies of target camera-viewer
Scanning dependencies of target video-viewer
[ 43%] Building CXX object utils/video/video-viewer/CMakeFiles/video-viewer.dir/video-viewer.cpp.o
[... cut ...]
[ 66%] Building CXX object CMakeFiles/jetson-inference.dir/c/imageNet.cpp.o
[ 67%] Building CXX object CMakeFiles/jetson-inference.dir/c/tensorNet.cpp.o
[ 68%] Building CXX object CMakeFiles/jetson-inference.dir/calibration/randInt8Calibrator.cpp.o
[ 68%] Building CXX object CMakeFiles/jetson-inference.dir/plugins/FlattenConcat.cpp.o
[ 69%] Linking CXX shared library aarch64/lib/libjetson-inference.so
[ 69%] Built target jetson-inference
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

mohanRajPartha, 2021-07-15:
> I was able to build from source succesfully. I will keep this open for now in case I have problems.

@DJT777  how the problem was solved, can you pls help ?

mohanRajPartha, 2021-07-15:
I found the problem for my error case, It was because of different python3 packages that was not properly installed.
To solve the problem,
1. uninstall python3 and libpython3 if it's not installed by default in Jetson Xavier NX. 
    I already had the python3.6.9 but i installed the python3.8 and this caused the problem.
2. Clone the repo and do the installation steps again.

Aldhanekaa, 2023-03-03:
> I found the problem for my error case, It was because of different python3 packages that was not properly installed. To solve the problem,
>
> 1. uninstall python3 and libpython3 if it's not installed [...]

Is your python3.6 still in your system, or you uninstall both (3.8 & 3.6, then install again)?

Aldhanekaa, 2023-03-03:
> Is your python3.6 still in your system, or you uninstall both (3.8 & 3.6, then install again)?

It seems like I face the same problem as you

closed 2023-04-27 by dusty-nv