dusty-nv/jetson-inference #793: Tensorflow to engine

lorenzo-caruso, 2020-11-11:
Hi guys,
First of all I want to thank you for the wonderful job you are doing, I tried the ssd mobile net following your tutorial and it worked like a charm. Now I am trying to use other networks not implemented but I have problems with NonMaximumSuppressionV2, because it is not supported by TRT so I can’t convert the tensorflow model to a TRT Engine, like you did in the tutorial. I would like to ask you how are you converting ssd mobile net to TRT Engine? Because I saw that after the tutorial a file .engine was saved in the project folder. Are you removing NMS and then applying transformations in parser?
If you are doing this, can I ask to you how are you removing those layers from pretrained networks? 

Thanks, Lorenzo

dusty-nv, 2020-11-24:
Hi @lorenzo-caruso , sorry for the delay - this was the tool used to convert the TensorFlow-based SSD-Mobilenet to TensorRT:

https://github.com/AastaNV/TRT_object_detection

Or if you are referring to the PyTorch-trained SSD-Mobilenet that the tutorial covers, I don't believe I had to remove anything from the network to get that working.  But the clustering is done in post-processing.

lorenzo-caruso, 2020-12-10:
Thank you very much!
Lorenzo

closed 2023-04-27 by dusty-nv