dusty-nv/jetson-inference #795: failed to initialize segnet

wuzhishiwo, 2020-11-13:


wuzhishiwo, 2020-11-13:
[image]

dusty-nv, 2020-11-13:
It appears that you have a space in `--network= fcn-alexnet-aerial-fpv-720p`

Can you try removing the space and trying `--network=fcn-alexnet-aerial-fpv-720p`

wuzhishiwo, 2020-11-14:
Thank you very much ,but I have remove the space, and the same problem appears, Are there other problems?my version is jetpack3.3, and I choose the L4T-R28.1 branch.

wuzhishiwo, 2020-11-14:
> It appears that you have a space in `--network= fcn-alexnet-aerial-fpv-720p`
> 
> Can you try removing the space and trying `--network=fcn-alexnet-aerial-fpv-720p`
Thank you very much ,but I have remove the space, and the same problem appears, Are there other problems?my version is jetpack3.3, and I choose the L4T-R28.1 branch.

closed 2023-04-27 by dusty-nv