dusty-nv/jetson-inference #796: Convert Tensoflow V2 model ( pb format) to UFF

Parham-khj, 2020-11-16:
I'm trying to use inference API (detectnet.py) to run an object detection model ( "efficientdet_d1" model which was trained by Tensorflow object detection API 2 on a local machine) on jetson Nano. I gather that UFF conversation (uff.from_tensorflow_frozen_model) doesn't support TF2 for direct conversion, so I'm following this pathway:

 ****PB format---> ONNX format ---> UFF**** 

I could successfully generate ONNX format. However, using "trtexec"  led to rasing below issue:

I was wondering if you have a solution.

Unsupported ONNX data type: UINT8 (2)
ERROR: ModelImporter.cpp:54 In function importInput:
[8] Assertion failed: convert_dtype(onnx_tensor_type.elem_type(), &trt_dtype)
[10/16/2020-14:30:38] [E] Failed to parse onnx file
[10/16/2020-14:30:38] [E] Parsing model failed
[10/16/2020-14:30:38] [E] Engine could not be created
&&&& FAILED TensorRT.trtexec # /usr/src/tensorrt/bin/trtexec --onnx=/media/pk/Data/Project/Project/Venice/Maet_stage1/efficientdet_d1_coco17_tpu-32/6class/ONNX/model_TRT.onnx

dusty-nv, 2020-11-16:
Can trtexec load the ONNX model instead?

Parham-khj, 2020-11-16:
> Can trtexec load the ONNX model instead?
> […](#)

That's the full output report:

[10/16/2020-14:30:37] [I] === Model Options ===
[10/16/2020-14:30:37] [I] Format: ONNX
[10/16/2020-14:30:37] [I] Model: /media/pk/Data/Project/Project/Venice/Maet_stage1/efficientdet_d1_coco17_tpu-32/6class/ONNX/model_TRT.onnx
[10/16/2020-14:30:37] [I] Output:
[10/16/2020-14:30:37] [I] === Build Options ===
[10/16/2020-14:30:37] [I] Max batch: 1
[10/16/2020-14:30:37] [I] Workspace: 16 MB
[10/16/2020-14:30:37] [I] minTiming: 1
[10/16/2020-14:30:37] [I] avgTiming: 8
[10/16/2020-14:30:37] [I] Precision: FP32
[10/16/2020-14:30:37] [I] Calibration: 
[10/16/2020-14:30:37] [I] Safe mode: Disabled
[10/16/2020-14:30:37] [I] Save engine: 
[10/16/2020-14:30:37] [I] Load engine: 
[10/16/2020-14:30:37] [I] Inputs format: fp32:CHW
[10/16/2020-14:30:37] [I] Outputs format: fp32:CHW
[10/16/2020-14:30:37] [I] Input build shapes: model
[10/16/2020-14:30:37] [I] === System Options ===
[10/16/2020-14:30:37] [I] Device: 0
[10/16/2020-14:30:37] [I] DLACore: 
[10/16/2020-14:30:37] [I] Plugins:
[10/16/2020-14:30:37] [I] === Inference Options ===
[10/16/2020-14:30:37] [I] Batch: 1
[10/16/2020-14:30:37] [I] Iterations: 10 (200 ms warm up)
[10/16/2020-14:30:37] [I] Duration: 10s
[10/16/2020-14:30:37] [I] Sleep time: 0ms
[10/16/2020-14:30:37] [I] Streams: 1
[10/16/2020-14:30:37] [I] Spin-wait: Disabled
[10/16/2020-14:30:37] [I] Multithreading: Enabled
[10/16/2020-14:30:37] [I] CUDA Graph: Disabled
[10/16/2020-14:30:37] [I] Skip inference: Disabled
[10/16/2020-14:30:37] [I] Input inference shapes: model
[10/16/2020-14:30:37] [I] === Reporting Options ===
[10/16/2020-14:30:37] [I] Verbose: Disabled
[10/16/2020-14:30:37] [I] Averages: 10 inferences
[10/16/2020-14:30:37] [I] Percentile: 99
[10/16/2020-14:30:37] [I] Dump output: Disabled
[10/16/2020-14:30:37] [I] Profile: Disabled
[10/16/2020-14:30:37] [I] Export timing to JSON file: 
[10/16/2020-14:30:37] [I] Export profile to JSON file: 
[10/16/2020-14:30:37] [I] 
----------------------------------------------------------------
Input filename:   /media/pk/Data/Project/Project/Venice/Maet_stage1/efficientdet_d1_coco17_tpu-32/6class/ONNX/model_TRT.onnx
ONNX IR version:  0.0.6
Opset version:    11
Producer name:    tf2onnx
Producer version: 1.8.0
Domain:           
Model version:    0
Doc string:       
----------------------------------------------------------------
WARNING: ONNX model has a newer ir_version (0.0.6) than this parser was built against (0.0.3).
Unsupported ONNX data type: UINT8 (2)
ERROR: ModelImporter.cpp:54 In function importInput:
[8] Assertion failed: convert_dtype(onnx_tensor_type.elem_type(), &trt_dtype)
[10/16/2020-14:30:38] [E] Failed to parse onnx file
[10/16/2020-14:30:38] [E] Parsing model failed
[10/16/2020-14:30:38] [E] Engine could not be created
&&&& FAILED TensorRT.trtexec # /usr/src/tensorrt/bin/trtexec --onnx=/media/pk/Data/Project/Project/Venice/Maet_stage1/efficientdet_d1_coco17_tpu-32/6class/ONNX/model_TRT.onnx

turowicz, 2021-01-20:
Same issue here @dusty-nv , seems like tensorflow object detection models aren't supported by tensorrt.

turowicz, 2021-01-20:
Same problem for tf 1.x

dusty-nv, 2021-01-20:
@turowicz this is the tool that was used to convert the TensorFlow to TensorRT:   https://github.com/AastaNV/TRT_object_detection

You might also try posting on the forums about it.

turowicz, 2021-01-20:
Unfortunately it doesn't support anything other than SSDs

closed 2023-04-27 by dusty-nv

turowicz, 2023-04-28:
Moving to Torch Vision was the fix.