dusty-nv/jetson-inference #797: detectnet on custom model (ssd-mobilenet-v2) got error cuTensor Error in executeCutensor: 7 (Internal cuTensor reformat failed)

KTOC, 2020-11-16:
Hi,

I retrained the detect model (ssd-mobilenet-v2) with my own dataset and successfully converted it to ONNX and now trying to run it on Jetson Nano 4G kit.

I used this command:

`./detectnet --model=/home/jetson/detect/mb2-ssd-lite.onnx --input-blob=input_0 --output-blob=boxes --output-cvg=scores --input_URI /home/jetson/detect/test6.mp4 --input-codec=mpeg4 --log-file=/home/jetson/detect/detectNet-Log.txt --debug --log-level=debug --headless` 

and got this error:

```
[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/jetson/detect/mb2-ssd-lite.onnx
[TRT]    ------------------------------------------------
[0;31m[cuda]      device not ready (error 600) (hex 0x258)
[0m[0;31m[cuda]      /home/jetson/detect/AI/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685
[0m[TRT]    Pre-Process   CPU   0.09969ms  CUDA   0.00000ms
[0;31m[cuda]      invalid resource handle (error 400) (hex 0x190)
[0m[0;31m[cuda]      /home/jetson/detect/AI/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685
[0m[TRT]    Network       CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.09969ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------
[gstreamer] gstDecoder -- recieved 1280x720 frame (1382400 bytes)
[0;31m[TRT]    ../rtSafe/cuda/cutensorReformat.cpp (352) - cuTensor Error in executeCutensor: 7 (Internal cuTensor reformat failed)
[0m[0;31m[TRT]    FAILED_EXECUTION: std::exception
[0m[0;31m[TRT]    failed to execute TensorRT context on device GPU
[0m[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/jetson/detect/mb2-ssd-lite.onnx
[TRT]    ------------------------------------------------
[0;31m[cuda]      device not ready (error 600) (hex 0x258)
[0m[0;31m[cuda]      /home/jetson/detect/AI/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685
[0m[TRT]    Pre-Process   CPU   0.05813ms  CUDA   0.00000ms
[0;31m[cuda]      invalid resource handle (error 400) (hex 0x190)
[0m[0;31m[cuda]      /home/jetson/detect/AI/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685
[0m[TRT]    Network       CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.05813ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------
```
If I remove the `--headless` option the video will be displayed but no inference on the video.

Full log is:

```
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for /home/jetson/detect/test6.mp4
[gstreamer] gstDecoder -- discovered video resolution: 1280x720  (framerate 30.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)3.1, profile=(string)main, width=(int)1280, height=(int)720, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=/home/jetson/detect/test6.mp4 ! qtdemux ! queue ! h264parse ! omxh264dec ! video/x-raw ! appsink name=mysink
[0;32m[video]  created gstDecoder from file:///home/jetson/detect/test6.mp4
[0m------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///home/jetson/detect/test6.mp4
     - protocol:  file
     - location:  /home/jetson/detect/test6.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      h264
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[0;31m[video]  videoOptions -- failed to parse output resource URI (null)
[... 200 lines cut ...]

[0;33m[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[0m[0;31m[TRT]    ../rtSafe/cuda/cutensorReformat.cpp (352) - cuTensor Error in executeCutensor: 7 (Internal cuTensor reformat failed)
[0m[0;31m[TRT]    FAILED_EXECUTION: std::exception
[0m[0;31m[TRT]    failed to execute TensorRT context on device GPU
[0m
[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/jetson/detect/mb2-ssd-lite.onnx
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.06438ms  CUDA   1.32266ms
[0;31m[cuda]      invalid resource handle (error 400) (hex 0x190)
[0m[0;31m[cuda]      home/jetson/detect/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685
[0m[TRT]    Network       CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.06438ms  CUDA   1.32266ms
[TRT]    ------------------------------------------------

[gstreamer] gstDecoder -- recieved 1280x720 frame (1382400 bytes)
[0;31m[TRT]    ../rtSafe/cuda/cutensorReformat.cpp (352) - cuTensor Error in executeCutensor: 7 (Internal cuTensor reformat failed)
[0m[0;31m[TRT]    FAILED_EXECUTION: std::exception
[0m[0;31m[TRT]    failed to execute TensorRT context on device GPU
```
I also attached the log file for your reference.
[detectNet-Log.txt](https://github.com/dusty-nv/jetson-inference/files/5544082/detectNet-Log.txt)

dusty-nv, 2020-11-16:
Hi @KTOC , how was your model trained? 

It appears to be using `mb2-ssd-lite`, which I have not tried that SSD-Lite variation with TensorRT (only SSD-Mobilenet trained from `jetson-inference/python/training/detection/ssd`)

KTOC, 2020-11-16:
Thank for your reply @dusty-nv. We trained the model using 
```
jetson-inference/python/training/detection/ssd/train_ssd.py
```
and converted to onnx using 

```
jetson-inference/python/training/detection/ssd/onnx_export.py
```
We also tried to run ssd-mobilenet model but got the same error.

Here is the model you can take a look.

```
https://drive.google.com/drive/folders/14cAzr011guUBjSm2Z-rBjPr4UbuQqaGv
```

The weird thing is we can run the onnx model both ssd-mobilenet and mb2-ssd-lite on PC using jetson-inference `./detectnet` or `detectnet.py after modifying the gst decode codec to fit the gst decode on PC.
I also tried the trtexe and it shows PASS and cuda-memcheck which shows following error:

```
[TRT]    detectNet -- number object classes:  4
[TRT]    detectNet -- maximum bounding boxes:  3000
[cuda]   cudaAllocMapped 1344000 bytes, CPU 0x100f68000 GPU 0x100f68000
[cuda]   cudaAllocMapped 64 bytes, CPU 0x100d6be00 GPU 0x100d6be00
[image] loaded '/home/jetson/detect/jetson-inference/build/aarch64/bin/images/city_0.jpg'  (1024x512, 3 channels)
[cuda]   cudaAllocMapped 1572864 bytes, CPU 0x1010c0000 GPU 0x1010c0000
========= Program hit cudaErrorInvalidValue (error 1) due to "invalid argument" on CUDA API call to cudaMemcpy2DAsync. 
=========     Saved host backtrace up to driver entry point at error
=========     Host Frame:/usr/lib/aarch64-linux-gnu/libcuda.so [0x2fd95c]
=========     Host Frame:/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcudart.so.10.2 (cudaMemcpy2DAsync + 0x18c) [0x49734]
=========
[TRT]    ../rtSafe/cuda/genericReformat.cu (1294) - Cuda Error in executeMemcpy: 1 (invalid argument)
[TRT]    FAILED_EXECUTION: std::exception
[TRT]    failed to execute TensorRT context on device GPU
[OpenGL] glDisplay -- set the window size to 1024x512
[OpenGL] creating 1024x512 texture (GL_RGB8 format, 1572864 bytes)
[cuda]   registered openGL texture for interop access (1024x512, GL_RGB8, 1572864 bytes)
[image] saved '/home/jetson/detect/result.jpg'  (1024x512, 3 channels)
========= Program hit cudaErrorInvalidResourceHandle (error 400) due to "invalid resource handle" on CUDA API call to cudaEventElapsedTime. 
[cuda]      invalid resource handle (error 400) (hex 0x190)
=========     Saved host backtrace up to driver entry point at error
[cuda]      /home/jetson/detect/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685
=========     Host Frame:/usr/lib/aarch64-linux-gnu/libcuda.so [0x2fd95c]
=========     Host Frame:./detectnet [0x39614]
=========

[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/jetson/detect/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   2.40922ms  CUDA 107.91261ms
[TRT]    Network       CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   2.40922ms  CUDA 107.91261ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[image] imageLoader -- End of Stream (EOS) has been reached, stream has been closed
detectnet:  shutting down...
detectnet:  shutdown complete.
========= ERROR SUMMARY: 2 errors

```

rebotnix, 2020-11-16:
Do you use Jetpack 4.4?

KTOC, 2020-11-16:
Yes I used Jetpack 4.4 @rebotnix

KTOC, 2020-12-08:
Close as the error come from the label file.

closed 2020-12-08 by KTOC

drostifrosti, 2021-01-06:
@KTOC: can you please elaborate on your solution. What do you mean by "label file"?
(btw I got a similar error "executeCutensor: 18 (Internal cuTensor reformat failed)" on an RTX)

ismet-dagli, 2021-04-07:
I have the same issue too. I am using the weights downloaded while building the repo.

dusty-nv, 2021-04-07:
> @KTOC: can you please elaborate on your solution. What do you mean by "label file"?

Perhaps they meant they had initially used the label file in their dataset, when it was expected to use the label file in their model folder.  The label file in the model folder has `BACKGROUND` class added to it by train_ssd.py

KTOC, 2021-04-07:
@dusty-nv @drostifrosti @ismet-dagli Please see this link (also posted by me). 

https://forums.developer.nvidia.com/t/detectnet-on-custom-model-ssd-mobilenet-v2-got-error-cutensor-error-in-executecutensor-7-internal-cutensor-reformat-failed/159618/6

As discussed in the link the error comes from the wrong number of classes in the label.txt, and if you use custom model you have to define these three parameters --input-blob --output-bbox --output-cvg

ismet-dagli, 2021-04-07:
Thank you so much for helping us with this. I will give it a try!